DomCR/ACadSharp #278: DwgWriter using  ACadVersion show his drawing was saved with a student Version

sigster, 2024-02-10:
Hi

If I use ACadVersion.AC1032 then I get = his drawing was saved with a student Version
but it is ok if I don't use ACadVersion

`    

    Const dwgFilePath22 = "Test1.dwg"
    doc.Header.Version = ACadVersion.AC1032

    Using writer As DwgWriter = New DwgWriter(dwgFilePath22, doc)
        writer.Write()
    End Using
End Sub`

Regards
Sigster

DomCR, 2024-02-12:
Hi @sigster 

This issue was fixed in this PR:
- https://github.com/DomCR/ACadSharp/pull/254

Please check that you have the latest branch on master or the latest Nuget Package.

If the still appears I'll open a branch to take a look.

sigster, 2024-02-14:
Finish to download master and check this

AC1024 = is OK
AC1032 = drawing was saved with a student Version

DomCR, 2024-02-16:
I've open a branch to take a look.

closed 2026-08-25 by DomCR