DomCR/ACadSharp #299: Linear Dimension: Wrong Measurement Value

mme1950, 2024-03-06:
**Describe the bug**
In a linear dimension the measurement we have the following values:
FirstPoint = {5801,705436813159,4162,800239003658}
SecondPoint = {5801,705436813158,3937,8002390044694}
The dimension line is vertical (Rotation = PI/2).
The value returned by Measurement is 6.

**Expected behavior**
Measurement = 225.

The measurement value should be the distance between `FirstPoint` and `SecondPoint` in the direction of the dimension line.
The direction of the dimension line is perpendicular to the extension line which goes from `SecondPoint` to `DefinitionPoint`. This should be the same as `Rotation`.

**Analyse**
The Measurement value is evaluated as follows:
```
		public override double Measurement
		{
			get
			{
				double rot = FirstPoint.AngleFrom(this.SecondPoint);
				return Math.Abs(FirstPoint.DistanceFrom(this.SecondPoint) * Math.Cos(this.Rotation - rot));
			}
		}
```
The `rot` value should not be considered.
This seems to work better:
`             return Math.Abs(FirstPoint.DistanceFrom(this.SecondPoint) * Math.Sin(this.Rotation));`

AutoCAD evaluates the measurement, why do we not use this value any more?

DomCR, 2024-03-06:
Hi @mme1950,

This seems to be a mistake during the implementation, maybe a C&P, not sure but the current code doesn't make sense, I'll open a branch and add some test to check property.

Thanks!

mme1950, 2024-03-06:
Hi @DomCR,

what is reason not to use the value provided by AutoCAD?

DomCR, 2024-03-06:
Is not dynamic, so if you create a dimension using `ACadSharp` the value always will be 0.

Also the measurement field is optional, that means that is ignored and calculated based on the given points and dimension type.

mme1950, 2024-03-06:
> Also the measurement field is optional, that means that is ignored and calculated based on the given points and dimension type.

This is an important aspect. Many entity properties provided by AutoCAD are calcuated from other values, e.g. in MultiLeader. In general the value should be calculated when entities are created.

closed 2024-03-08 by DomCR