DomCR/ACadSharp #304: Get Attributes

Gora1000, 2024-03-07:
Hello

how is the best way to get the attributes from a Block in a Layout?

DomCR, 2024-03-12:
Hi @Gora1000 

To get the attributes for an instance of a block you have to get all the Inserts in the model, like:

```C#
CadDocument doc = DwgReader.Read(path);
foreach (Insert insert in doc.Entities.OfType<Insert>())
{
	//Do something with: insert.Attributes;
}
```

Let me know if this helps.

Gora1000, 2024-03-14:
Hi DomCr,

thaks for your answer. With doc.Entities you get all blocks of type insert in the model space, but not from layout. For exsample I added a test.dwg to my answer. 
[[Test.zip](https://github.com/DomCR/ACadSharp/files/14600868/Test.zip)]

I you open the file, you will find 3 Blocks in the Model space and in each layout the same one. In test.dwg two Layouts. If you use
`CadObjectCollection<Entity> ent= doc.Entities;`

you will get only three blocks from model space, but i need the Attributes from the Block in Layout.

If it should work, and i can change attribute values on the block in a layout. It is possible to save the changes at the same dwg file?

DomCR, 2024-03-14:
Each Layout has it's own Block associated to it, which is the one that contains all the entities, you can iterate throw all layouts using:

```C#
foreach (Objects.Layout layout in doc.Layouts)
{
      BlockRecord block = layout.AssociatedBlock;
      foreach (Insert insert in block.Entities.OfType<Insert>())
      {
	      //Do something with: insert.Attributes
      }
}
```

This way you will get all the block instances (Insert) in the Layout that you want.

You can save the file using `DwgWriter` but keep in mind that `ACadSharp` doesn't guarantees the integrity of the resulting file, so I would not recommend to override the file, instead save it with a different name.

Let me know if this helps.