DomCR/ACadSharp #313: ACAD_TABLE Implementation

DomCR, 2024-03-18:
### Discussed in https://github.com/DomCR/ACadSharp/discussions/312

<div type='discussions-op-text'>

<sup>Originally posted by **tardisconsole** March 18, 2024</sup>
Hi,

I'm writing an import tool that simply needs to parse some content text contained in a table inside a DWG.  
Using AcadSharp, I can see virtually everything except the table data.  Reading some other discussions relating to ACAD_TABLE, it seems this has not yet been implemented.  

Doing a foreach through the document entities, I have found BlockRecord entries that match the majority of the content text contained in the table, but there are cases where the content text in the table is an empty string.  These entries are not included in the BlockRecord list when reading in the DWG file.  Because of this, it's impossible to tell where the empty strings are when running through the list.

Will ACAD_TABLE support be available soon? Just a list of entities with grid row/column references and the content text would essentially solve my issue.
Or alternatively, is there a way of including the empty BlockRecord entries (if they exist)?

Thank you.</div>

DomCR, 2024-11-26:
Hi @NeilWBruce,

The Table has a lot of properties and subclasses but the dxf information is limited, I wrote a start but it was a big time consuming so I stopped the work.

I'll retake this one shortly.

NeilWBruce, 2024-11-26:
Thank you,  I've managed to work around it for now by exporting the data to CSV but it be great to grab it straight from the Table but yeah the standard looks very complex.

closed 2024-12-07 by DomCR