DomCR/ACadSharp #315: autocad cant read library dwg/dxf writing output

mamaorha, 2024-03-19:
Autocad cant read library dwg/dxf writing output.

To reproduce:
1. download [Drawing.zip](https://github.com/DomCR/ACadSharp/files/14646367/Drawing.zip) and extract it
2. run the following code loading the Drawing.dwg that was inside the zip

`using ACadSharp;
using ACadSharp.Entities;
using ACadSharp.IO;
using CSMath;

CadDocument doc = DwgReader.Read(@"Drawing.dwg");

using (DxfWriter writer = new DxfWriter(@"Drawing.dxf", doc, false))
{
    writer.Write();
}

using (DwgWriter writer = new DwgWriter(@"Drawing.dwg", doc, false))
{
    writer.Write();
}`

expected a readable file in autocad but the output is much smaller then original dwg and fail in autocad.

DomCR, 2024-03-20:
Hi @mamaorha,

I've test your file, here are the results, DXF:

[image]

DWG:

[image]

With the DWG I needed to recover the file but it seems that it could be opened by Autocad.

I've noticed that your file has a dynamic block, this are not fully supported by `ACadSharp`, which may be the cause for the loss of information in the file. I'll use your file to try to fix the issue and improve the reader/writer.

mamaorha, 2024-03-20:
Thank you for addressing it so quickly, i truely appreciate the work you have done here.
im guessing that with time the missing read/write parsers will preserve more data.
indeed autocad offers to recover but still a lot of the data is lost in that transition (you can see the file shrinked from 144kb to 36kb).

Nico3201975, 2024-05-10:
Same issue here. Wandering has this bug been fixed?

DomCR, 2024-05-10:
Hi @Nico3201975 

This particular issue was resolved, the file could be opened in Autocad with no major issue, if you are referring to the size, this is a work in progress to be solved.

In any other case, if you have a particular issue with a file, it would be good if you could provide an example of what are you trying to achieve or the file that is not working.

Nico3201975, 2024-05-11:
Hi @DomCR 

Thanks for the quick reply.

I just updated to the newest release version.
Now I am on the stage of inserting a summary info into a file created by autocad.
Ingoring the size of the file part, a recovery prompt shows up each time when openning it.
Is there a way to removing this notification by any chance?

DomCR, 2024-05-13:
That notification usually appears for a badly written entity or object, after pressing recover Autocad usually prompts a log with what has been fixed and which entities or objects have been affected, if you can provide the logs I can take a look of what is the cause of it.

Also if you could provide an example of the output file it will really help with the problem because I could take a more detailed look at the issue.

Nico3201975, 2024-05-14:
Hi DomCR,

So basically, I followed the example to call DwgReader to retrieve a dwg file attr, then set comment section, then output to a new dwg file by calling DwgWriter.

Here is added comment dwg file:
[log&file.zip](https://github.com/DomCR/ACadSharp/files/15302320/log.file.zip)

DomCR, 2024-05-16:
I've tested your file, this is the code that I'm using:

```C#
	CadDocument doc = DwgReader.Read(test, this._dwgConfiguration, this.onNotification);

	doc.SummaryInfo.Comments = "HELLO THIS IS MY COMMENT";

	DwgWriter.Write(path, doc, this.onNotification);
```

I'm guessing that this is what you want to achieve, if your case is more complex or you are changing some other structure, please let me know to make sure that I can properly create a test case.

The resulting file seems healthy, this is the only popup that I get form Autocad:

[image]

Which is completely normal and it doesn't recover any errors.

Checking the changes:

[image]

Are you using the Nuget package or the repository?

DomCR, 2024-05-16:
[image]

another test with more data, the Hyperlink hasn't been saved though.

Nico3201975, 2024-05-17:
Hi DomCR,

Appreciate for the testing.

what I wrote is pretty much same as what you provided above.

DwgReaderConfiguration _dwgConfiguration = new DwgReaderConfiguration
{
	KeepUnknownEntities = true,
	Failsafe = false
};
CadDocument doc = DwgReader.Read(strDwgFilePath, _dwgConfiguration);
doc.SummaryInfo.Comments = strDwfFileComment;
using (DwgWriter writer = new DwgWriter(strNewDwgFinal, doc))
{
//writer.OnNotification += NotificationHelper.LogConsoleNotification;
writer.Write();
}

DwgReader.Read() actually reads a file originally created by Autocad2022, then adds a comment, then after that creating as a new dwg file.

Somehow, my autocad2022 shows the prompt below:
[image]

And here is the prompt file:
[123dwg.zip](https://github.com/DomCR/ACadSharp/files/15344988/123dwg.zip)

If nothing looks suspected about above snippet, I might look over, and debug myself one more time carefully.
Thanks for the response.