DomCR/ACadSharp #327: True Color: R and B interchanged

mme1950, 2024-03-23:
**Describe the bug**
It seems that, when reading or writing colours, the red and the blue component are interchanged.

**To Reproduce**
Create a multileader with kind-of blue leader line.
Debug the Color R G B values: we get kind-of orange.

**Expected behavior**
It should be the RGB as selected in AutoCAD.

**Solution**
Change line 39 in `DwgStreamReaderAC18 `to:
`color = new Color(arr[2], arr[1], arr[0]);`
and change `DwgStreamWriterAC18` accordingly.

**Related Refactoring**
We have an implementation of `ReadCmColor` in `DwgStreamReaderAC18` that can read a colour index or a true colour.
The implementation in `DwgStreamReaderBase` is for older versions. It just reads a colour index

But:
In `DwgMergedReader.ReadCmColor` the base implementation is called `if (!(this._mainReader is DwgStreamReaderAC18))`.
Otherwise, the same code as the implementation of `ReadCmColor` in `DwgStreamReaderAC18` follows.
If I did not miss anything the implementation in `DwgMergedReader` can be reduced to.
```
	public Color ReadCmColor()
	{
		return this._mainReader.ReadCmColor();
	}
```

mme1950, 2024-03-23:
[20240322_ml_with_style.zip](https://github.com/DomCR/ACadSharp/files/14731021/20240322_ml_with_style.zip)

mme1950, 2024-03-24:
BTW:
What does this comment tell?
				//CECOLOR:
				//3221225472
				//0b11000000000000000000000000000000
				//0b1100_0000_0000_0000_0000_0000_0000_0000 --> this should be ByLayer
				//0xC0000000

3221225472 = 0xC0000000 = 0b11000000000000000000000000000000

0b1100_0000_0000_0000_0000_0000_0000_0000 indicates a true color, otherwise this bit must be set:
0b0000_0001_0000_0000_0000_0000_0000_0000.
So this is not ByLayer.

But it would be interesting to know what the meaning of the other bits of the byte 3 are:
We read 
* 0b1100_0010 a 0b1100_0011 from DWG files
* 0b1100_0000 a 0b1100_0001 from DWG files

DomCR, 2024-03-25:
Hi @mme1950,

I'll open a branch to fix the order of the bytes.

About the `DwgMergedReader.ReadCmColor` is implemented this way because for versions after AC1018 have the text stream at the end, which means that we need to read the color book name (by now is not supported) using the _textStream and not the _mainStream, there is a comment at the begining of the method:

```C#
//To read a color in this version file needs to access to the
//string data section at the same time.
```

Is not ideal but does the trick.

> //CECOLOR:

I did some tests changing the values in the Current Entity Color and I've noticed that some encoding seem to change for this specific value, that's why that comment is there.

mme1950, 2024-03-25:
Hi @DomCR,
so `DwgStreamReaderAC18.ReadCmColor` is never called, but the identical code following the `if (!(this._mainReader is DwgStreamReaderAC18)) ...`.

mme1950, 2024-03-25:
> //CECOLOR:
If the value is 3221225472 = 0xC0000000 = 0b11000000000000000000000000000000 we have a true color and we reach the else branch.
This value will not be read as ByLayer. If it should it must be handled explicitly.

closed 2024-03-26 by DomCR

DomCR, 2024-03-27:
Sorry, forgot about that one, I'll add a test too.

thanks!

DomCR, 2024-03-27:
> What is the reason that in `DwgMergedReader.ReadCmColor`
> `DwgStreamReaderAC18.ReadCmColor` is never called, but the identical code following the
> ` if (!(this._mainReader is DwgStreamReaderAC18))` ....

That if is to call the `ReadCmColor` for versions that are lower than AC1018, in that case we only need the index, so the "duplicated" code has to be override.

mme1950, 2024-03-27:
Hi @DomCR,

would not this code do the same:
```
	public Color ReadCmColor()
	{
		return this._mainReader.ReadCmColor();
	}
```

If _mainReader is a base class of `DwgStreamReaderAC18`, the code that writes only a color index is called.
The code in `DwgStreamReaderAC18` can write both.
May be I  missed the point.

DomCR, 2024-03-28:
The reason for the duplication is this part of the method:

```C#
	string colorName = string.Empty;
	//RC: Color Byte(&1 => color name follows(TV),
	if ((id & 1) == 1)
	{
		colorName = this.ReadVariableText();
	}

	string bookName = string.Empty;
	//&2 => book name follows(TV))
	if ((id & 2) == 2)
	{
		bookName = this.ReadVariableText();
	}
```

The `ReadVariableText` uses the `_textReader` because the text is stored at the end of the entity, so it  cannot be handled by the main reader, that's why there is a duplication.

Let me know if I resolved your question.

mme1950, 2024-03-28:
Oh, I understand. **this** in `DwgMergedReader` is clearly not the **this** in `DwgStreamReaderAC18`. Is the code in `DwgStreamReaderAC18` called and can it work at all?