DomCR/ACadSharp #332: How can I modify the font size in a dwg file after reading it?

lp956, 2024-03-27:
This is my code, seeking help. Thank you！！！

using DwgReader reader = new(filename: "D:\\robot\\Learning\\WSA P3-2 layout.dwg");
//using DxfReader reader = new DxfReader(filename: "D:\\robot\\test444.dxf");
CadDocument doc = reader.Read();
foreach (var s in doc.TextStyles) {
    s.Height = 0.1;
    s.LastHeight = 0.1;
}
foreach (var s in doc.DimensionStyles) { 
    s.TextHeight = 0.1;
    s.ArrowSize = 0.1;
    s.CenterMarkSize= 0.1;
    s.TickSize = 0.1;
}
//foreach (var entity in doc.Entities.Where(x=>x.ObjectType == ObjectType.MTEXT)) {            
//    foreach (var d in entity.Document.TextStyles) {
//        d.Height = 1;
//        d.LastHeight = 0;
//        d.Width = 1;
//    }
//    foreach (var d in entity.Document.DimensionStyles) {
//        d.TextHeight = 0.1;                
//    }
//}        
using DxfWriter dxfWriter = new("D:\\robot\\test44454.dxf", doc, false);        
dxfWriter.Write();

lp956, 2024-03-27:
It's not effective for me to modify the Height in this way, so I don't know where to modify it. I'm worried

DomCR, 2024-03-27:
Hi @lp956, 

In your code you are modifying the `TextStyle` but not the text entities so the ones that are in the drawing will not be affected, to change the size from the text in the drawing you should change the Height in the entities:

```C#
	foreach (MText mtext in doc.Entities.OfType<MText>())
	{
		mtext.Height = 20;	//Change the value of the hight
	}
```
 
Here are the changes in the file that I've tested:

Test file:
[image]

Modified file:
[image]

Note that this code will only change the height for `MText` entities not `TextEntity`, you can use the interface `IText` to apply the change to all of them regarding if they are `MText` or `TextEntity`.

Let me know if this helps.

lp956, 2024-03-27:
I have successfully solved the font size issue according to the plan you provided. Thank you very much ！

lp956, 2024-03-27:
> Hi @lp956,
> 
> In your code you are modifying the `TextStyle` but not the text entities so the ones that are in the drawing will not be affected, to change the size from the text in the drawing you should change the Height in the entities:
> 
> ```cs
> 	foreach (MText mtext in doc.Entities.OfType<MText>())
> 	{
> 		mtext.Height = 20;	//Change the value of the hight
> 	}
> ```
> 
> Here are the changes in the file that I've tested:
> 
> Test file: ![init_state](https://private-user-images.githubusercontent.com/46492487/317230592-f1a632b4-08e2-4713-8825-7bd5e5b3a810.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE1NDcyOTksIm5iZiI6MTcxMTU0Njk5OSwicGF0aCI6Ii80NjQ5MjQ4Ny8zMTcyMzA1OTItZjFhNjMyYjQtMDhlMi00NzEzLTg4MjUtN2JkNWU1YjNhODEwLkpQRz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzI3VDEzNDMxOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI0N2NiMDE0MmRhYzI2ODliOGRjZjRiODg1YjkxNTczYzk2NzYxODdiOWU2ZDEzZjg1NWY0OTYzNmU0MjM4ZGImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.tUP13gFyfHacz1aDc8Ux2uZkzxjHX_B7JDOUvn-e66U)
> 
> Modified file: ![end_state](https://private-user-images.githubusercontent.com/46492487/317230646-7597db1a-24eb-4cc8-bc33-40d03d32498d.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE1NDcyOTksIm5iZiI6MTcxMTU0Njk5OSwicGF0aCI6Ii80NjQ5MjQ4Ny8zMTcyMzA2NDYtNzU5N2RiMWEtMjRlYi00Y2M4LWJjMzMtNDBkMDNkMzI0OThkLkpQRz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzI3VDEzNDMxOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWYzZjMzZjRlMmM0YWI0ZGQ5YjI3ZTBhM2QyY2JiMmFiMDc0MWMyNTFmYjk4NmM4MjJhNjFlOWMzOGJlZDVhNGMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.tWOAOo7F4TGd4cPPnUOVqY6LubSvqJd3WBWl_1l5ml0)
> 
> Note that this code will only change the height for `MText` entities not `TextEntity`, you can use the interface `IText` to apply the change to all of them regarding if they are `MText` or `TextEntity`.
> 
> Let me know if this helps.

I have successfully solved the font size issue according to the plan you provided. Thank you very much ！

DomCR, 2024-03-27:
Anytime!!

Closing this issue