DomCR/ACadSharp #343: Optimization of DWG File Parsing Speed

LianYangCn, 2024-04-17:
https://github.com/DomCR/ACadSharp/blob/abf321c024ff53b8ea748dc57fcde36b93c62e46/ACadSharp/Objects/CadDictionary.cs#L204-L205

Those two lines of code slow down the file parsing speed.
I have a DWG file with approximately 180,000 dictionary entries.
Direct parsing takes around 300 seconds, but if those two lines are commented out, parsing takes about 40 seconds.

I am not sure what side effects those two lines of code may bring, 
but I have tested them with 10 DWG files that I usually work with, 
and they all work fine.

DomCR, 2024-04-17:
Hi @LianYangCn 

That's actually a good point, I'll open a branch to take a look at this.

Thanks!

LianYangCn, 2024-04-18:
Hi @DomCR

I must say this library is very good to use.

Thanks a ton

closed 2024-04-22 by DomCR