DomCR/ACadSharp #345: Saving a DXF file not working

rafalklebba, 2024-04-24:
**Describe the bug**
I've just started using this library and I have a problem while adding a very simple functionality. I want to read a file and save it without any changes. It reads the file and saves it but the saved file is a lot smaller and cannot be opened in AutoCAD, when I try the program crashes.

**To Reproduce**
`
CadDocument doc = DxfReader.Read(file.OpenReadStream());

doc.Header.Version = ACadVersion.AC1018;
using (DxfWriter writer = new DxfWriter(guid + ".dxf", doc, false))
{
    writer.Write();
}
`
This code reads and saves the file but the resulting file is a smaller eg. original is 319KB and the one saved by this library is 60KB.

**Expected behavior**
I would like to see the original file saved correctly. In the end I want to be able to read a file, add entities to it and return it back to the user.

**Screenshots**
I'm including the crashLog from AutoCAD that I get when I try to open the created file.
[image]

Is there anything that I'm doing wrong?

DomCR, 2024-04-24:
I've been looking at the issue, if is related to the layout, I've merged a fix for an error that I've found in the DxfWriter.

- https://github.com/DomCR/ACadSharp/pull/346

Let me know if this fixes the issue.

rafalklebba, 2024-04-24:
Hi, thank you for such a quick response. I'll take a look at this as soon as I can, probably tomorrow and I'll let you know if the problem is fixed.

As for the file that I'm using to test it, it is this one:
[test_file.dxf.zip](https://github.com/DomCR/ACadSharp/files/15100188/test_file.dxf.zip)

rafalklebba, 2024-04-24:
I managed to test it now and it seems to be working correctly for the 2 files that I tested which didn't work before. Thank you very much for the quick update.

DomCR, 2024-04-25:
Glad to help! if you have any more issues please report them so the project can improve.

closed 2024-04-25 by DomCR