DomCR/ACadSharp #349: DwgWriter write error

KiisMyGun, 2024-05-14:
# Describe the bug
Error saving original file as new file：System.ArgumentOutOfRangeException:“Index was out of range. Must be non-negative and less than the size of the collection. Arg_ParamName_Name”
# To Reproduce
[test.zip](https://github.com/DomCR/ACadSharp/files/15309287/test.zip)
After decompression, read this file and save it as a new file
```
var input = "D:\\Users\\26295\\Documents\\KLW单梯土建图1-销售技术部测试-KLW-1250-1-L1-4-15.dwg";
var output = "D:\\Users\\26295\\Documents\\test9.dwg";

CadDocument doc = DwgReader.Read(input);

//New document to transfer the entities
CadDocument transfer = new CadDocument();
doc.Header.Version = doc.Header.Version;

//Nove the entities to the created document
List<ACadSharp.Entities.Entity> entities = new List<ACadSharp.Entities.Entity>(doc.Entities);
foreach (var item in entities)
{
    ACadSharp.Entities.Entity e = doc.Entities.Remove(item);
    transfer.Entities.Add(e);
}

//Save the document
using (DwgWriter writer = new DwgWriter(output, transfer))
{
    writer.Write();
}
```
# Expected behavior
## Based on the original dwg file. I hope to generate 50 new files in bulk.
- Modify several objects through multiple parameters, such as stretching/text modification, etc.
- Then generate new files in bulk.
- If this object is associated with other objects, if one is changed, other objects can also be updated together
**Screenshots**
[image]
# Additional context
[test.zip](https://github.com/DomCR/ACadSharp/files/15309287/test.zip)

DomCR, 2024-05-16:
Hi @KiisMyGun,

The issue seems to be when the Hatch is been written, I'll open a branch to take care of this issue.

Thanks for reporting the bug.

KiisMyGun, 2024-05-20:
Hi @DomCR 

Thanks for the quick reply.

closed 2024-05-23 by DomCR