DomCR/ACadSharp #353: get error when use memory stream to read files

farshadvl, 2024-06-03:
hi 
I use acadsharp to read dwg files and do some process like dxf files on it.
my code:
`using (MemoryStream ms = new MemoryStream())
{

    using (DxfWriter writer = new DxfWriter(ms, doc, false))
    {
        writer.OnNotification += onNotification;
        writer.Write();
    }

    DxfFile dxfFile = DxfFile.Load(ms);
}

it get error : Cannot access a closed Stream.

what should I do?

Revan1985, 2024-06-03:
Hi @farshadvl 
are you sure you want to read a file?
Class DxfWriter is used to write, you should use class DxfReader.

DomCR, 2024-06-03:
Hi @farshadvl,

As @Revan1985 pointed out your piece of code seems to be incomplete for us to help, but for what you are saying you are trying to read information from a closed `Stream`, if that's the case try to create a copy of the same stream:

```C#
	using (MemoryStream ms = new MemoryStream())
	{
		CadDocument doc = new CadDocument();

		using (DxfWriter writer = new DxfWriter(ms, doc, false))
		{
			writer.CloseStream = false;
			writer.Write();
		}

		MemoryStream s = new MemoryStream(ms.ToArray());

		using(DxfReader reader = new DxfReader(s))
		{
			CadDocument output = reader.Read();
		}
	}
```

I'm not sure if this is what you are trying to achieve, let me know if this helps.

DomCR, 2024-06-03:
I've also noticed that the writer closes the stream even if the `CloseStream` flag is set to false, I'll open a branch to fix it, in the mean time the code that I've provided should help.

closed 2024-07-18 by DomCR