DomCR/ACadSharp #357: Stack Overflow Exception When Reading Large File After Updating from 2.2.0-beta to 2.3.1-beta

BrunoSam66, 2024-06-09:
**Describe the bug**
After updating from version 2.2.0-beta to 2.3.1-beta, reading the same file now throws a Stack Overflow exception.

**To Reproduce**
Just reading a file .dxf with 64 MB and it takes too long to read the file (30 minutes or more).

**Expected behavior**
Using the previous version, I was able to read the file in less than 3 minutes.

**Screenshots**
[image]
```

Stack overflow.
at ACadSharp.CadObjectCollection`1[[System.__Canon, System.Private.CoreLib, Version=8.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]].GetEnumerator()
at System.Linq.Enumerable+WhereEnumerableIterator`1[[System.__Canon, System.Private.CoreLib, Version=8.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]].MoveNext()
at System.Linq.Enumerable.Any[[System.__Canon, System.Private.CoreLib, Version=8.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]](System.Collections.Generic.IEnumerable`1[System.__Canon])
at ACadSharp.Parser.Shapes.BlockIdentifier+c__DisplayClass8_0.<GetInsertsByType>b__0(ACadSharp.Entities.Insert)
at System.Collections.Generic.List`1[[System.__Canon, System.Private.CoreLib, Version=8.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]].ForEach(System.Action`1[System.__Canon])
at ACadSharp.Parser.Shapes.BlockIdentifier.GetInsertsByType(ACadSharp.CadDocument, ACadSharp.Parser.Enums.BlockType)
at ACadSharp.Parser.Parser.<Build<BuildList>b__GetTargetBlock_2_1(System.Nullable`1[KCSMath.XY>, ACadSharp.CadDocument)
at ACadSharp.Parser.Parser.<Build<BuildList>b__GetTargetBlock_2_2(ACadSharp.CadDocument, ACadSharp.Entities.Insert, System.Collections.Generic.List`1[ACadSharp.Entities.LwPolyline>, ACadSharp.Parser.ParserOperations.Tree)
```

**Additional context**
I cannot share the file or reproduce an equivalent one, but it has 3000 INSERTS and some LwPolylines interacting with them.
[image]
[image]

DomCR, 2024-06-09:
Hi @BrunoSam66 

About the Stack Overflow exception it seems that is caused in your code, after the method `GetTargetBlock` there is a infinite recursion between the methods `BuildTree` and `SearchNewBranch`, without more information about what methods or properties are you calling is hard to know what is happening in the code, could you give me some clue of what is happening in these methods?

About the time, I'm checking the PR's from the release but I don't find any major modification in the `DxfReader` besides adding the entities `Shape` and `Image` that messes with the time to read, I'll try to replicate the issue by adding stress tests using big files.

I'll keep looking on what caused this decrease on the performance.

DomCR, 2024-06-11:
I've tested the issue with a file of 115MB for both versions the 2.3.1 has a bottleneck issue in the builder, which is affecting the dwg and dxf, after a quick fix this was the result:

`9 seconds for 2.2.0`
`12 seconds for 2.3.1`

I've opened a branch to try to improve these results.

Thanks for reporting the issue!

closed 2024-06-11 by DomCR