DomCR/ACadSharp #363: DXF file not loading

rafalklebba, 2024-06-11:
Hi,
I have a problem where I try to load a DXF file. On version 2.3.1-beta I get this error: "Object reference not set to an instance of an object." and on version 2.3.2-beta I get this error: "An item with the same key has already been added. Key: 0".

My code:
CadDocument doc = new CadDocument();
doc = DxfReader.Read(file.OpenReadStream());

I'm sure that the 'file' variable is not null and has the correct data.

Is there anything I'm doing wrong? What can I do to fix it?

DomCR, 2024-06-12:
Hi @rafalklebba,

I've been seeing this exception on files that don't specify the Dxf version or have an irregular format that doesn't match the documentation, it seems that the Dxf format is quite "free" with it's format.

I already have a branch to fix this issue, I'll link it to the issue.

If you could provide a file that triggers this error that would help the development, I could use it in the test cases.

DomCR, 2024-06-12:
In the mean time you can try to use the method `ReadEntities` in the `DxfReader` it will try to read the entities in the file, depending on the format of the file you will be able to retrieve them, check this discussion for more information:

- https://github.com/DomCR/ACadSharp/discussions/354#discussioncomment-9735666

closed 2024-06-14 by DomCR