DomCR/ACadSharp #368: How to read and write large DWG files?

Sring65, 2024-06-15:
Hello
200MB以上的dwg读取缓慢，内存爆满，有什么方法能够方便读写？
Reading DWG files over 200MB is slow, and memory usage is high. Are there any methods to facilitate reading and writing?

```
namespace ACadSharp.Examples
{
	class Program
	{
		//const string _file = "../../../../samples/sample_AC1032.dwg";
		const string _file = "../../../../samples/sample_AC1018_Bigfile.dwg"; // >200MB
               ......
                doc = reader.Read(); //卡住 slow
```

DomCR, 2024-06-23:
Hi @Sring65,

For DWG there is no way around, you need to read the whole file, the information is scattered in sections and they are dependent from each other.

For the performance issue, make sure that you are using the latest master, recently I've merged this PR which increases the performance of the reader, but I haven't tested this kind of huge files myself, the rating right now is a DWG file of 24MB and a DXF file of 115MB, they are tested to be read in at least 10 seconds, I'll try to replicate your error using a bigger file to see if I can find a bottleneck but this might be a limitation regarding the file size.

- https://github.com/DomCR/ACadSharp/pull/360

Sring65, 2024-07-25:
hi @DomCR 
    经过测试常规200M的dwg文件打开文件80s左右，但是包含xdata数据的200M dwg在读取的时候卡住了
After testing, it was found that a conventional 200MB DWG file takes about 80 seconds to open, but the 200MB DWG file containing xdata gets stuck during the reading process.

Sring65, 2024-07-26:
[image]
https://1drv.ms/f/s!Am5OJstFAyTsgQGzryGlAKVPb--d?e=EEn3yD

DomCR, 2024-08-03:
Hi @Sring65,

I've been looking at your file, I opened using Autocad, it took more 3 minutes to open and exhausted all my 32GB RAM, `ACadSharp` does not have a visual interface so it should be faster and consume less RAM, but giving the size and the time that Autocad took to open the file is quite a challenge to optimize it.

Thanks for providing a file to test the library, I'll try to use it as an example to optimize the library, but I cannot take a clear action to fix or increase performance right now.

closed 2024-09-14 by Sring65