DomCR/ACadSharp #381: DwgWriter fails due to summaryinfo being null

Nubbit, 2024-07-16:
**Describe the bug**
I have hundreds of DWG files made by a company with a custom (Autodesk authorized) solution where the summaryinfo in the CadDocument is null. DwgReader has no issue with this but when writing a new DWG file based of the CadDocument the writeSummaryInfo() method of DwgWriter fails due to the SummaryInfo being null. 

**Expected behavior**
The expected behavior would be that ReadSummaryInfo() initializes a new CadSummaryInfo object when the retrieved value from the DWG file is null as is with a newly initialized CadDocument.

**Screenshots**
[image]

**Additional context**
I understand/expect that the DWG files are out of spec but it would be nice that DwgWriter/writeSummaryInfo() doesn't fail because of it. Sadly I am not allowed to share the DWG file.

Also, what an awesome project! Keep up the great work!

DomCR, 2024-07-17:
Hi @Nubbit,

Your issue may be caused by this code:

```C#
	//Older versions than 2004 don't have summaryinfo in it's file
	if (this._fileHeader.AcadVersion < ACadVersion.AC1018)
		return null;
```

Older version do not have SummaryInfo, but you are right it should not be null by default.
Opening a branch to fix it.

Thanks for your support!

closed 2024-07-17 by DomCR

Nubbit, 2024-07-17:
I wasn't paying attention to the file version but you are absolutely right! Thank you for the quick fix.