DomCR/ACadSharp #389: Encoding Issue

jyoubnnggg13, 2024-07-23:
**Describe the bug**
trying to read the dwg and write it to dxf file, the file contains Korean text content.
the content is displayed as ' ? ' and some sort of special chars.

**Expected behavior**
Need matched encoding 

**Screenshots**
[image]

jyoubnnggg13, 2024-07-23:
And I think this doc has right pagecode
[image]

DomCR, 2024-07-24:
Hi @jyoubnnggg13 

I'll open a branch to take a look.

Can you provide an example file to test it?

Thanks!

jyoubnnggg13, 2024-07-25:
@DomCR  Thanks for quick reply !

I attached the file I used in .zip ( because of not supported file extension )
+ I found another dwg that contains Korean text and has same codepage in its header ( kcs5601 ) brought  me result I expected.

Thanks :) 
[test1.zip](https://github.com/user-attachments/files/16368655/test1.zip)

closed 2024-07-25 by DomCR