DomCR/ACadSharp #391: HELP WANTED: How to convert cadDocument into a pdf file.

KyleeChangg, 2024-07-23:
As the title suggests, I currently have a cadDocument but cannot find in the documentation how to convert this document into a pdfFile. Any help would be appreciated, thanks!

DJGosnell, 2024-07-23:
It is a complicated process and to my knowledge, there is no public .NET solution for this, which is why my company needed to create this system from scratch.  Our system uses SKIA as the [PDF creation engine](https://learn.microsoft.com/en-us/dotnet/api/skiasharp.skdocument.createpdf?view=skiasharp-2.88).  

The process is as follows: Convert all the simple objects to their equivalent SKIA counterparts. Render the blocks into `SKPicture`s then apply the transformation matrix to them to place them into the correct location.  Rendering the text is also a bit tricky as you have to parse MText values.  I have an old version of our parser located in this [PR](https://github.com/DomCR/ACadSharp/pull/94).

DomCR, 2024-07-24:
Hi @KyleeChangg 

The current state of the library doesn't allow to export the file in PDF, I might take a look after the v1 is released.

DomCR, 2024-09-03:
It not functional yet but I started to take a look:

https://github.com/DomCR/ACadSharp.Pdf