DomCR/ACadSharp #398: Hatch Transparency Value Not Updating in DWG file

KyleeChangg, 2024-07-28:
`private static Hatch CreateHatchForPolyline(IPolyline polyLine)
		{
			var transparency = new Transparency()
			{
				Value = 50
			};

			var hatchBoundaryPath = new Hatch.BoundaryPath()
			{
				Flags = BoundaryPathFlags.External, // Set appropriate flags as needed
				Entities = new List<Entity> { polyLine as Entity } // Add the polyline directly
			};

			// Create the hatch and set its properties
			var hatch = new Hatch()
			{
				Paths = new List<Hatch.BoundaryPath> { hatchBoundaryPath },
				IsSolid = true,
				Transparency = transparency,
				IsAssociative = true,
			};

			return hatch;
		}`

**Describe the bug**
In the code above, I created a function that creates hatches in relation to polyline entities with the hatch transparency value set to 50%.

[image]

These hatches show up fine in the DWG file however, the hatch transparency value is reset to "ByLayer" and the 50% transparency does not appear. 

[image]

KyleeChangg, 2024-08-04:
@DomCR Any pointers for where I can find when the hatch values are written to the document?

DomCR, 2024-08-04:
Hi @KyleeChangg,

I've been looking at this issue, I'll need to open a branch to explore it further, for the DXF the transparency value is not implemented, and for the DWG it seems that there is an error in the method `WriteEnColor` the transparency value doesn't seem to take any effect on the entity written.

Thanks for the report!

closed 2024-08-04 by DomCR