DomCR/ACadSharp #399: Some entities don't seem to be imported it

Lukeor, 2024-08-01:
Hello,

First of all my knowledge of CAD files isn't the best and so apologies if this issue is more of a user error & lack of understanding from my side.

I have attached a CAD file which I am trying to read in.

[X - MASTERPLAN.zip](https://github.com/user-attachments/files/16455422/X.-.MASTERPLAN.zip)

When viewing the file in DWG TrueView, I am able to see the red outlines of buildings throughout the map. Here is a screenshot of how they appear in the bottom-right corner of the view:

[image]

However when I try to import the file into my own software using ACadSharp the red outlines don't seem to get imported:

[image]

The red outlines seem to belong to a layer called 'UP-BLDG-RESi' but I can't seem to locate these particular entitites in the ACadSharp cad document. Here is an example of one of the entities after being selected in QCAD:

[image]

Are you able to see if these do get read in by ACadSharp okay or whether I am simply misunderstanding how to access them?

Many thanks for your help and don't hesitate to ask if you require some more information.

Luke

DomCR, 2024-08-03:
Hi @Lukeor,

I tested the file using the latest version on master, here is the result for the same area:

[image]

It seems that the lines are there, are you using the Package or the repo? if you are using the Package I can create a release.

About the file, it has a lot of Proxy entities, which are not currently implemented, I'll try to use the file as an example but getting information about how to read proxies is quite difficult.

Thanks for the report! this helps the project.

Lukeor, 2024-08-05:
Hey DomCR,

Many thanks for your reply.

I am using the Nuget Package in my code, I am using 2.4.0-beta.

No worries about the CAD file. I hope it helps!

Kind regards,

Luke