DomCR/ACadSharp #407: 保存DWG文件后实体颜色不对

xmap2008, 2024-08-06:
打开一个有颜色实体的DWG文件，即可保存，实体颜色变了。
源图
[image]

保存后
[image]

源文件
[t.zip](https://github.com/user-attachments/files/16504206/t.zip)

代码
        static void Main(string[] args)
        {
            string path = @"D:\tmp\t.dwg";
            CadDocument doc = DwgReader.Read(path);
            using (DwgWriter writer = new DwgWriter(@"D:\tmp\t1.dwg", doc))
            {
                writer.Write();
            }
        }

DomCR, 2024-08-06:
hi @xmap2008,

Could you translate the issue to English.

Thanks!

xmap2008, 2024-08-06:
The entity color is incorrect in the saved DWG file

DomCR, 2024-08-07:
I've found an issue in the Color Struct, opening a branch to fix it.

Thanks!

closed 2024-08-07 by DomCR