DomCR/ACadSharp #409: GDAL reads a DXF file shows 0 features Which created by ACadSharp

Fsystem, 2024-08-07:
**Describe the bug**
QGIS/GDAL reads a DXF file shows 0 features Which created by ACadSharp(read dwg and convert to dxf),But shows ok in Acme CAD Converter,and if I use Acme CAD Converter to convert dwg to dxf ,this dxf can be read in QGIS/GDAL

**To Reproduce**
[test.zip](https://github.com/user-attachments/files/16517856/test.zip)

Steps to reproduce the behavior or an example of the code used.

**Expected behavior**
GDAL reads a DXF file Which created by ACadSharp  should shows none-zero features 

**Screenshots**
[image]

**Additional context**
Add any other context about the problem here.
If the issue is related to an specific dwg/dxf file and if able, attach that file to the issue of an equivalent with the same error.

DomCR, 2024-08-08:
Hi @Fsystem

I've imported a dxf to QGIS, the result seem to work fine:

[image]

I've tested 2 versions of dxf, AC1018 and AC1032, both seem to be in the project.

Could you provide an example of a file that is failing.

Thanks.

Fsystem, 2024-08-08:
> test.zip

the test.zip is the dwg  that I  want  to convert to dxf

DomCR, 2024-08-10:
It seems that the writer and reader have some error in regarding of the encoding of the file, I'll open a branch to fix it.

closed 2024-08-14 by DomCR