DomCR/ACadSharp #412: Dictionary entries with key ACAD_ENHANCEDBLOCK nod read anymore

mme1950, 2024-08-09:
We have a DWG with a `BlockRecord` an `XDictionary` that has two enties ACAD_SORTENTS, ACAD_ENHANCEDBLOCK. This can be seen in the DXF. The second entry ACAD_ENHANCEDBLOCK was read previously, but is now missing.

This is important to read dynamic blocks.

DomCR, 2024-08-09:
Hi @mme1950,

Can you provide an example file? I've been looking if I have one with that entry but didn't find any.

Thanks!

mme1950, 2024-08-09:
Hi @DomCR,
I found the problem:
* The entry points to an `EvaluationGraph` object which is implemented in our branch [in nanoLogika/ACadSharp](https://github.com/nanoLogika/ACadSharp/tree/20240619_mme_%236-support-dynamic-blocks-read-evaluation-graphs-and-block-visibility-parameters)
* `EvaluationGraph` must no be a `NongraphicalObject`.

Changing the base class of `EvaluationGraph` solves the problem.

So it no problem in ACadSharp.

mme1950, 2024-08-09:
No ACadSharp problem.

closed 2024-08-09 by mme1950