DomCR/ACadSharp #417: Hatches not writing correctly

gktval, 2024-08-12:
**Describe the bug**
I am trying to write a Polygon from NetTopologySuite (https://github.com/NetTopologySuite/NetTopologySuite) to a hatch. It seems to write, but when I load it into AutoCad, it does not seem to read correctly. I am not very well versed in auto cad files, so I am hopeful someone can help me pinpoint where I am going wrong.

**To Reproduce**
Here is the code I am using to create the hatch:

```

Hatch hatch = new Hatch();

// loop through the geometries in the polygon
for (int n = 0; n < drw.Geometry.NumGeometries; n++)
{
    Geometry geom = drw.GetBasicGeometryN(n);
    if (geom.OgcGeometryType == OgcGeometryType.Polygon)
    {
        List<Hatch.BoundaryPath.Line> edges = new List<Hatch.BoundaryPath.Line>();

        //edge
        for (int i = 0; i < geom.Coordinates.Length - 1; i++)
        {
            Hatch.BoundaryPath.Line edge = new Hatch.BoundaryPath.Line();
            edge.Start = new CSMath.XY(geom.Coordinates[i].X, geom.Coordinates[i].Y);
            edge.End = new CSMath.XY(geom.Coordinates[i + 1].X, geom.Coordinates[i + 1].Y);
            edges.Add(edge);
        }

        //add the connection back to the start point
        Hatch.BoundaryPath.Line connectionEdge = new Hatch.BoundaryPath.Line();
        connectionEdge.Start = new CSMath.XY(geom.Coordinates[geom.Coordinates.Length - 1].X, geom.Coordinates[geom.Coordinates.Length - 1].Y);
        connectionEdge.End = new CSMath.XY(geom.Coordinates[0].X, geom.Coordinates[0].Y);
        edges.Add(connectionEdge);

        Hatch.BoundaryPath path = new Hatch.BoundaryPath();
        path.Edges.AddRange(edges);
        hatch.Paths.Add(path);
    }

This successfully creates a hatch, but the hatch entity is not read by other AutoCad programs

**Expected behavior**
This issue occurs with both dwg and dfx files. I have also tried creating ```Hatch.BoundaryPath.Polyline``` instead of ```Hatch.BoundaryPath.Line```. there is no write methods for a ```Polyline```. I created one, but still could not get a solution to work. Not sure if I am missing a Flag somewhere or what I need to do to write the Hatch properly. Any help is appreciated.

DomCR, 2024-08-13:
Hi @gktval 

It seems that the hatch are there but without a pattern defined they need the flag `IsSolid` set to true, this may solve the issue and show the Hatch in Autocad.

I'll open a branch to implement the polyline boundary.

Thanks for the report.

gktval, 2024-08-14:
That was it, I needed to set the IsSolid. It works now.

Another change I made is the capability to set the vertices for Polylines. Right now, the vertices property is private set, so there is no way to create a Polyline (only read one internally).  So for example, I added this code to Polyline2D:

```
/// <summary>
/// Sets the vertices for Polyline2D
/// </summary>
/// <param name="vertices"></param>
public void SetVertices(Vertex2D[] vertices)
{
	this.Vertices.AddRange(vertices);
}
```

then a similar method for Polyline3D.

Not sure if this is something you are interested in or not...

closed 2024-08-19 by DomCR