DomCR/ACadSharp #421: Dxf reading further than end of file or stream

jhvjnr, 2024-08-15:
**Describe the bug**
When reading **diamond-bin.dxf** into a CadDocument, I get the following error:

```
System.IO.EndOfStreamException: Unable to read beyond the end of the stream.
   at System.IO.BinaryReader.ReadByte()
   at ACadSharp.IO.DXF.DxfBinaryReaderAC1009.readCode()
   at ACadSharp.IO.DXF.DxfStreamReaderBase.ReadNext()
   at ACadSharp.IO.DxfReader.getReader()
   at ACadSharp.IO.DxfReader.Read()
   at ACadSharp.IO.DxfReader.Read(Stream stream, NotificationEventHandler notification)
```

**To Reproduce**

I run the following code:

`doc = ACadSharp.IO.DxfReader.Read(memoryStream);`

Providing diamond-bin.dxf as a MemoryStream
Find here 3 files:
diamond-bin.dxf -> throws error
diamond.dxf -> this file works
contours.dxf -> throws error shown lower down in this issue
[Examples.zip](https://github.com/user-attachments/files/16623795/Examples.zip)

**Expected behavior**
Expected a successful read of the file like I see in other programs

**Screenshots**
The eventual output should look like this as seen in another cad program:

[image]

**Additional context**
I've also attached diamond.dxf, a text representation of the same data that does work.

Also attached is contours.dxf, which fails with this error when opening:
```
ACadSharp.Exceptions.DxfException: Invalid dxf code with value 0, at line 2037859.
   at ACadSharp.IO.DXF.DxfStreamReaderBase.transformValue(GroupCodeValueType code)
   at ACadSharp.IO.DXF.DxfStreamReaderBase.ReadNext()
   at ACadSharp.IO.DXF.DxfTextReader.ReadNext()
   at ACadSharp.IO.DxfReader.getReader()
   at ACadSharp.IO.DxfReader.Read()
   at ACadSharp.IO.DxfReader.Read(Stream stream, NotificationEventHandler notification)
```

I believe these errors may be related.

Thank you for reading this!

DomCR, 2024-08-16:
Hi @jhvjnr,

The files that are failing seem to be irregular, they don't have any header or definitions about the version or format.

I'll open a branch to try to fix it.

Thanks for reporting the bug.

jhvjnr, 2024-08-16:
> Hi @jhvjnr,
> 
> The files that are failing seem to be irregular, they don't have any header or definitions about the version or format.
> 
> I'll open a branch to try to fix it.
> 
> Thanks for reporting the bug.

Great thank you, I will try find out how they were created too. I have opened diamond-bin.dxf and contours.dxf successfully in an application that uses the ODA Teigha libraries

closed 2024-08-18 by DomCR