DomCR/ACadSharp #432: Cannot write Chinese text correctly

rogure, 2024-08-26:
When I used it, I found that I could not write Chinese text into MTEXT. When I saved it to DWG file, the Chinese MTEXT displayed "?????". Symbol. How can this be resolved?

DomCR, 2024-08-30:
Hi @rogure,

I've tested this code and write the result into a DWG:

[image]

[image]

I'm using the master branch of this repo and seems to work.
Are you using the repo or the package?

If you are using the package can release an update.

CatarinaVCh, 2025-02-06:
Used the repository and when I saved to DWG file, the Chinese TEXT and MTEXT displayed "?????"
How can this be resolved?

DomCR, 2025-02-07:
Hi @CatarinaVCh,

I've tested this again and it seems that only `MText` adapts the Chinese characters correctly. 

To solve this you need to create a `TextStyle` compatible with them, like so:

```C#
TextStyle style = new TextStyle("custom");
style.Filename = "romans.shx";
style.BigFontFilename = "chineset.shx";
```

Remember to add the style to your entities:

```C#
TextEntity text = new TextEntity();
text.Value = "我的短信";
text.Style = style;
```

Here is more information about the solution:
https://www.autodesk.com/support/technical/article/caas/sfdcarticles/sfdcarticles/Chinese-text-shows-as-a-question-mark-in-AutoCAD.html

Hope this helps.

CatarinaVCh, 2025-02-07:
Hi, @DomCR ,
Thanks for your help, but unfortunately, this doesn't work for me.
I don't know why, but any text not in Latin is displayed in DWG as '???'

DomCR, 2025-02-07:
Which software are you using? 
Can you add the characters for a file that has not been created with `ACadSharp`?
Can you share the code that you are using to generate the file?

CatarinaVCh, 2025-02-07:
Using Window 10, Autodesk Autocad 2019 and Microsoft Visual Studio Community 2022 (17.6.5)
After opening the received file I can insert any symbols

My code:

            CadDocument doc = new CadDocument();
            Line line = new Line
            {
                StartPoint = CSMath.XYZ.Zero,
                EndPoint = new CSMath.XYZ(100, 100, 0)
            };

            TextEntity text = new TextEntity();
            TextStyle style = new TextStyle("custom");
            style.Filename = "romans.shx";
            style.BigFontFilename = "chineset.shx";

            text.Value = "jhfja hfjfg 我的短信";
            text.Style = style;
            text.Height = 50;
            text.Color = new ACadSharp.Color(new byte[] { 200, 20, 100 });
            text.InsertPoint = new CSMath.XYZ(10, 10, 0); 
                                                        
            string initialName1 = "blocks_layer";
            string initialName2 = "text_layer";
            Layer layer1 = new Layer(initialName1);
            Layer layer2 = new Layer(initialName2);

            doc.Layers.Add(layer1);
            doc.Layers.Add(layer2);

            line.Layer = layer1;
            doc.Entities.Add(line);

            text.Layer = layer2;
            doc.Entities.Add(text);

            VPort vport = doc.VPorts["*Active"];
            if (vport != null)
            {
                vport.Center = new CSMath.XY(0.0, 0.0);
                vport.ViewHeight = 2202.0730586005643;
                vport.ViewMode = ViewModeType.FrontClippingZ;
                vport.UseDefaultLighting = true;
            }

            var desktopPath2 = Environment.GetFolderPath(Environment.SpecialFolder.DesktopDirectory);
            string savePath2 = Path.Combine(desktopPath2, "2.dwg");

            using (DwgWriter writer = new DwgWriter(savePath2, doc))
            {
                writer.Write();
            }

CatarinaVCh, 2025-02-08:
Thank you very much for checking the code, but it doesn't work for me)))
I checked the fonts, they are there)))
The style of the text corresponds to those installed programmatically, but the symbols are displayed as '???'
Applying this style to other text fields in Autocad itself works, but not to those created with ACadSharp
It seems that some encoding is applied during the saving process, and then '???' remain '???' in any encoding and fonts
I don't know, maybe Autocad was installed with errors...

DomCR, 2025-02-08:
You can try to change the file encoding then:

```C#
CadDocument.Header.CodePage = "GB2312"; //Chinese simplified code page (GB2312)
```

Maybe for your version is needed.

Another test you can try is to save a document where the characters work and check the `TextStyle` and the `Header` encoding.

CatarinaVCh, 2025-02-24:
Thanks, change the file encoding is what I needed

closed 2025-11-21 by DomCR