DomCR/ACadSharp #443: How read dwg Entity through handles like in taigha?

Sring65, 2024-09-10:
How read dwg Entity through handles like in taigha?
怎么通过句柄读取数据
taigha code
`                               
Database db = new Database(false, true);
db.ReadDwgFile(dwgFilePath, FileOpenMode.OpenTryForReadShare, false, DwgPassword, true);
 Handle handle = new Handle((long)reader["Handle"]); //创建索引sqlite.db存储句柄数据范围等数据
                                db.TryGetObjectId(handle, out oid);
                                if (oid == null || oid == ObjectId.Null)
                                    continue;
                                using (DBObject dbObj = oid.GetObject(OpenMode.ForRead))
                                {
                                    if (dbObj is Entity)
                                    {
                                        using (Entity entity = dbObj as Entity)`

DomCR, 2024-09-10:
Hi @Sring65,

I'm not familiar with taigha, but maybe what you are looking for is this method:

```C#
CadDocument.GetCadObject(ulong handle)
```

Returns any `CadObject` in the document by it's handle.

Let me know if it helps.

Sring65, 2024-09-14:
hi @DomCR 
taigha的ReadDwgFile方法， 打开dwg文件并没有解析数据打开文件比较快，通过句柄访问才解析对应数据，这样对只处理dwg文件内部分数据有优势，节省了加载所有数据时间，ACadSharp打开数据只读取句柄的设置吗？
The ReadDwgFile method of Taigha opens DWG files without parsing the data, which makes the file opening process faster. The data is only parsed when accessed through the handle, offering an advantage for processing only specific data within the DWG file, as it saves time on loading all the data. Does ACadSharp also have a setting to read data by only accessing the handles?

DomCR, 2024-09-18:
I'll open a branch to take a look.

The problem that I see is that the reading of the file is sequential and some information may be lost in the process.

closed 2024-09-23 by Sring65