DomCR/ACadSharp #452: Drawing file is not valid

amirj0914, 2024-09-17:
Autocad gives the error after wrriting DWG file with this simple code : 

`

            string dwgFilePath = @"G:/output.dwg";

            CadDocument doc = new CadDocument(ACadVersion.AC1024);

            //Create a point located in (10, 10, 0)
            Point pt = new Point
            {
                Location = new CSMath.XYZ(10, 10, 0)
            };

            //Create a line from the origin to the point (5, 5, 0)
            Line line = new Line
            {
                StartPoint = CSMath.XYZ.Zero,
                EndPoint = new CSMath.XYZ(5, 5, 0)
            };

            doc.Entities.Add(pt);
            doc.Entities.Add(line);

            DwgWriter writer = new DwgWriter(dwgFilePath, doc);
`
After opening the exported file with Autocad 2017 :

[image]

thaiguer, 2024-09-18:
Hello, @amirj0914

Make shure you are using the latest version: dotnet add package ACadSharp --version 1.0.0

I created a new file with the following code, and I had success opening it with BricsCad.

`using ACadSharp;
using ACadSharp.Entities;
using ACadSharp.IO;

//old line:
//string dwgFilePath = @"G:/output.dwg";

//new line:
string dwgFilePath = @"C:\dev\output.dwg";

CadDocument doc = new CadDocument(ACadVersion.AC1024);

//Create a point located in (10, 10, 0)
Point pt = new Point
{
    Location = new CSMath.XYZ(10, 10, 0)
};

//Create a line from the origin to the point (5, 5, 0)
Line line = new Line
{
    StartPoint = CSMath.XYZ.Zero,
    EndPoint = new CSMath.XYZ(5, 5, 0)
};

doc.Entities.Add(pt);
doc.Entities.Add(line);

DwgWriter writer = new DwgWriter(dwgFilePath, doc);

//new line
writer.Write();`

There are a few diferences to your code. First, the dwgFilePath, the way you wrote it, seems to be incorrect. And in the end, you forgot to use the method "Write", from the class DwgWriter.

thaiguer, 2024-09-18:
Hello, @amirj0914

Make shure you are using the latest version: dotnet add package ACadSharp --version 1.0.0

I created a new file with the following code, and I had success opening it with BricsCad.

`Hello, @amirj0914

Make shure you are using the latest version: dotnet add package ACadSharp --version 1.0.0

I created a new file with the following code, and I had success opening it with BricsCad.

There are a few diferences to your code. First, the dwgFilePath, the way you wrote it, seems to be incorrect. And in the end, you forgot to use the method "Write", from the class DwgWriter.`

There are a few diferences to your code. First, the dwgFilePath, the way you wrote it, seems to be incorrect. And in the end, you forgot to use the method "Write", from the class DwgWriter.

`using ACadSharp;
using ACadSharp.Entities;
using ACadSharp.IO;

//old line:
//string dwgFilePath = @"G:/output.dwg";

//new line:
string dwgFilePath = @"C:\dev\output.dwg";

CadDocument doc = new CadDocument(ACadVersion.AC1024);

//Create a point located in (10, 10, 0)
Point pt = new Point
{
    Location = new CSMath.XYZ(10, 10, 0)
};

//Create a line from the origin to the point (5, 5, 0)
Line line = new Line
{
    StartPoint = CSMath.XYZ.Zero,
    EndPoint = new CSMath.XYZ(5, 5, 0)
};

doc.Entities.Add(pt);
doc.Entities.Add(line);

DwgWriter writer = new DwgWriter(dwgFilePath, doc);

//new `line`
writer.Write();`

amirj0914, 2024-09-18:
Thank you so much for your quick reply. 
The problem has been resolved. Exporting DWG files without AutoCAD was an ambition of mine.
I need two help : 
- Exportin DWG for Autocad AC1021 (version 2007)
- Displying DWG on a Form without needing to Autocad.

DomCR, 2024-09-18:
Hi @amirj0914,

Right now `ACadSharp` can export in this [versions](https://github.com/DomCR/ACadSharp?tab=readme-ov-file#compatible-dwgdxf-versions).

Unfortunately the version `AC1021` is not available due this version has an specific encryption, it has been resolved for the reader but reversing the code for the writer is quite a job.

For `AC1027` there is an unknown error in the code that I have to find.

I'll open a branch to keep in mind to add the feature to write files in `AC1021` but I cannot guarantee that this will be implemented soon.

amirj0914, 2024-09-21:
Thank yo so much, 
My target is **AC1021** for writing dwg file , I will appricate you if you resolve the problem for this version.

closed 2026-11-03 by DomCR