DomCR/ACadSharp #458: Diferences on the file created by AcadSharp

thaiguer, 2024-09-25:
Hello,

I used the following code to copy entities from one file to a new one:

```
using ACadSharp;
using ACadSharp.IO;

Console.WriteLine("Hello, World!");

string sourceFile = @"C:\dev\tablet.dwg";
string targetFile = @"C:\dev\just a drawing222.dwg";

using (DwgReader reader = new DwgReader(sourceFile))
{
    CadDocument doc = reader.Read();
    CadDocument newDoc = new CadDocument();

    foreach (var entity in doc.Entities)
    {
        doc.Entities.Remove(entity);
        newDoc.Entities.Add(entity);
    }

    using (DwgWriter writer = new DwgWriter(targetFile, newDoc))
    {
        writer.Write();
    }
}
```

I attached the source and the target file. [files.zip](https://github.com/user-attachments/files/17130376/files.zip)
The problem is that there are a few differences and some missing entities in the target file.

Problem 1: Hatches missing inside Block References

Source:
[image]

Target:
[image]

Problem 2: Different color on a ByLayer Entity

Source:
[image]

Target:
[image]

Problem 3: OLE 2 Entity missing

Source:
[image]

Target:
[image]

DomCR, 2024-09-26:
Hi @thaiguer,

The first problem seems to be related to the `Hatch` entities, I guess that is something related to the patterns, I have to take a look.

The second problem I've found an error in the `Layer` clone method.

I'll open 2 branches to fix this issues.

Thanks for the report!

thaiguer, 2024-09-27:
Hello,
The problem 2 seems to be fixed, thanks for the correction.
Although, the Problem 1: **Hatches missing inside Block References**, is still happening.

Take a look at a image from the source file:
[image]

And take a look at a image from the targe file:
[image]

[Uploading samples.zip…]()

DomCR, 2024-10-04:
RULESURF still not fix.

closed 2024-10-04 by DomCR

thaiguer, 2024-10-04:
> Hello, The problem 2 seems to be fixed, thanks for the correction. Although, the Problem 1: **Hatches missing inside Block References**, is still happening.
> 
> Take a look at a image from the source file: ![image](https://private-user-images.githubusercontent.com/107768648/371518448-a70ec26b-a964-4eb5-9028-7322e6a15ac3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjgwNjQyNzMsIm5iZiI6MTcyODA2Mzk3MywicGF0aCI6Ii8xMDc3Njg2NDgvMzcxNTE4NDQ4LWE3MGVjMjZiLWE5NjQtNGViNS05MDI4LTczMjJlNmExNWFjMy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAwNFQxNzQ2MTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hYWRmOTdlMzRkYjhiNmI3ZTU1ODYyMTQ2NDY4ZTdhN2Y2YzQzZDg3MzE3NTY1ZjczZTQ5Nzk1MDRiOGVjOGJmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.zPdWv0l4gd4X83J4jLsajtAMXqzjtouFFWYH6IPb3Y8)
> 
> And take a look at a image from the target file: ![image](https://private-user-images.githubusercontent.com/107768648/371518608-c537db77-d9fc-4746-adc6-d9e8cbe877bb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjgwNjQyNzMsIm5iZiI6MTcyODA2Mzk3MywicGF0aCI6Ii8xMDc3Njg2NDgvMzcxNTE4NjA4LWM1MzdkYjc3LWQ5ZmMtNDc0Ni1hZGM2LWQ5ZThjYmU4NzdiYi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAwNFQxNzQ2MTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mZWNkZTNiNDc0MDFhOGI4M2Q3ZTU1OTAwYjhiMDdjMzQ4Yjg1ODZlYzIwY2IzYmJiYmJlNjAxZjNmNGU0NTQ3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.K1WwYk94yUDVy73masBt_tBOJ0rGreKaBC_9YWHKp3Y)
> 
> Uploading samples.zip…

The Hatch problem seems to be solved.