DomCR/ACadSharp #471: DwgReader.Read not reading Polylines in Example file

jhvjnr, 2024-10-11:
**Describe the bug**
When reading Example.dwg into a CadDocument, the CadDocument contains no entities, even though the file contained Polylines

[image]

**To Reproduce**

I run the following code:

doc = ACadSharp.IO.DwgReader.Read(memoryStream);

Providing Example.dwg as a MemoryStream

[Example.zip](https://github.com/user-attachments/files/17340926/Example.zip)

**Expected behavior**
Expected a successful read of the file like I see in other programs

**Screenshots**
The eventual output should look like this as seen in another cad program:

[image]

DomCR, 2024-10-12:
Hi @jhvjnr,

The file that you send me doesn't have any polyline, all the entities seem to be proxy entities which are not stored in the file unless specified by the reader configuration. 

Unfortunately `ACadSharp` is not capable of storing the geometric data yet, so you will only get the DxfClass of the proxy entity.

jhvjnr, 2024-10-14:
Hi @DomCR thank you for having a look at this and the feedback.

I think that at least some data must be stored in the file somehow as it is >6MB and I was sent the file as is, in isolation. Opening it in Autodesk TrueView shows this:

[image]

It looks like layers C-TOPO and HS2-00002BJS5-HFS-C01_Breaklines have polylines.

Maybe I have a misunderstanding? Should I be looking elsewhere than cadDocument.Entities?

Hopefully it is something simple like this!

Thanks,
jhvjnr

DJGosnell, 2024-10-14:
In addition to simple drawing entities, AutoCAD can contain many different types of much more complex entities known as Proxy Entities.  Essentially they are a blob of data (like an image for example) that are to be read a certain way by the viewer and those methods have never been made public (as far as I know) for any the many proxy entities.  If you find some documentation about how they are read (and written) that would be a path forward to be able to parse the data.  Otherwise, reading this data would be an arduous task of trial and error by modifying the data in the CAD file and seeing how the output changes.  This is also assuming the data in the proxy entity doesn't completely change every time, like in the case of certain encodings, compressions or encryptions.

DomCR, 2024-10-14:
You can set the flag `KeepUnknownEntities` in the reader configuration to true so it keeps all these entities as unknown, but as @DJGosnell said, they will not contain any geometric data, only the `DxfClass` that identifies the proxy and the handle.

closed 2025-11-11 by DomCR