DomCR/ACadSharp #473: Multileader attached to a line - wrong coordinates of ContextData.BasePoint

blahetal, 2024-10-15:
**Describe the bug**
Hello, I am using the library to enumerate closed polylines, free floating lines and multileaders in a dwg file. Multileaders are attached to the corners of polylines and to the start or end point of lines. One multileader can be attached to only one entity (polyline or line). I can get X/Y coords of all polylines, lines and multileaders. My aim is to know which multileader is connected to which entity (polyline or line) by comparing X/Y coords of corners (polyline) or start/end point (line) with X/Y of multileaders. 

In the dwg file (attached), there is a line which has one attached multileader to its start/end point. But when I enumerate all the entities and output their X/Y coords, there is no match between multileader X/Y coord and line start/end point. The values are different. For getting multileader X/Y coord I am using the property `ContextData.BasePoint.X/Y`.

Then there is a polygon which has another multileader attached to its corner and their (the corner and multileader) coords match to the pixel. Could you check it please.

**To Reproduce**
```cs
using ACadSharp.Entities;
using ACadSharp.IO;
using ACadSharp;

CadDocument doc = DwgReader.Read(@"test.dwg");

foreach (Entity item in doc.Entities)
{
    if (item is not null && item is LwPolyline polyline)
    {
        Console.WriteLine("Polyline");
        foreach (LwPolyline.Vertex vertex in polyline.Vertices)
        {
            Console.WriteLine($"\tX = {vertex.Location.X}, Y = {vertex.Location.Y}");
        }
    }
    else if (item is not null && item is Line line)
    {
        Console.WriteLine("Line");
        Console.WriteLine($"\tstart X = {line.StartPoint.X}, start Y = {line.StartPoint.Y}, end X = {line.EndPoint.X}, end Y = {line.EndPoint.Y}");
    }
    else if (item is not null && item is MultiLeader multiLeader)
    {
        Console.WriteLine("multiLeader");
        Console.WriteLine($"\tX = {multiLeader.ContextData.BasePoint.X}, Y = {multiLeader.ContextData.BasePoint.Y}, Text = {multiLeader.ContextData.TextLabel}");
    }
}

Console.ReadLine();
```

**Expected behavior**
Coords of mltileaders attached to start/end point or corners of polygon should have correct coords (same as the corresponding corner or start/end point).

**Screenshots**
This is an output of the code snippet above. The mutlileader with text `vnitrni cara` is attached to the `Line` yet their coords do not match.

[image]

Here is a screenshot of the dwg with circled multileader attached to an end point of the line.

[image]

**Additional context**
-

DomCR, 2024-10-17:
Hi @blahetal,

The connection points that you are looking for are stored in `ContextData.LeaderRoots[].Lines[].Points[]`.

You will need to iterate throw a couple of lists due the multileaders support multiple lines/points to be attached to.

Let me know if this helps!

blahetal, 2024-10-17:
thank you, that was the correct property to get the coords from

closed 2024-10-17 by blahetal