DomCR/ACadSharp #474: ARES Commander Cad

ChandrasekarML, 2024-10-16:
While opening the output drawing, an error drawing file occurs and requires recovery. 
Input drawing file images are available, but output drawing file images are invisible.

sample code:
using ACadSharp.Examples;
using ACadSharp.IO;
using ACadSharp.Tables.Collections;
using ACadSharp.Tables;
using System;
using System.Diagnostics;
using System.Linq;
using ACadSharp.Entities;
using System.Xml.Linq;

namespace ACadSharp.Examples
{

    public class Program
    {
        static void Main(string[] args)
        {

            void OnNotification(object sender, NotificationEventArgs e)
            {
                Console.WriteLine($"Notification: {e.Message}");
            }

            const string sourceFile = "C:\\Users\\Balaji.t\\Videos\\INPUT5\\FILE\\INPUT.dwg";
            const string targetFile = "C:\\Users\\Balaji.t\\Videos\\INPUT5\\FILE\\OUTPUT.dwg";      
           
            CadDocument doc;

            using (DwgReader reader = new DwgReader(sourceFile))
            {
                doc = reader.Read();

            }
            

            void WriteDwg(string targetFile, CadDocument doc)
            {
              
                using (DwgWriter writer = new DwgWriter(targetFile, doc))
                {                 
                    writer.OnNotification += OnNotification;
                    writer.Write();
                }
            }

            WriteDwg(targetFile, doc);

        }
    }
}
[FILE.zip](https://github.com/user-attachments/files/17394796/FILE.zip)

DomCR, 2024-10-17:
Hi @ChandrasekarML,

I can see that the file contains a dynamic block and an instance of that one.

Dynamic blocks are not yet supported and my guess is that is messing with the output file because the writer does not distinguish between dynamic and normal.

I'll take a look to see if I find a workaround to make this compatible.

Thanks for your report!

ChandrasekarML, 2024-10-17:
While opening the output drawing file, the drawing file requires recovery notification to appear in version 1.0.1.
The output drawing file opens properly in beta version 2.0.1. 
But both version sample codes are the same.

What you tried sample code in below:

using ACadSharp;
using ACadSharp.Entities;
using ACadSharp.IO;
using System.Reflection.Metadata;
using System.Xml.Linq;

class Program
{
    static void Main(string[] args)
    {

        string dwgFilePath = "C:\\Users\\Balaji.t\\Videos\\INPUT5\\Test.dwg";

        CadDocument doc = new CadDocument();

        void OnNotification(object sender, NotificationEventArgs e)
        {
            Console.WriteLine($"Notification: {e.Message}");
        }

        //Create a line from the origin to the point (5, 5, 0)
        Line line = new Line
        {
            StartPoint = CSMath.XYZ.Zero,
            EndPoint = new CSMath.XYZ(5, 5, 0)
        };

        //doc.Entities.Add(pt);
        doc.Entities.Add(line);

        doc.Header.Version = ACadVersion.AC1032;

        using (DwgWriter writer = new DwgWriter(dwgFilePath, doc))
        {
            writer.OnNotification += OnNotification;
            writer.Write();
        }

    }
}
[FILE 1.zip](https://github.com/user-attachments/files/17410411/FILE.1.zip)

ChandrasekarML, 2024-10-25:
What is the status of the previous chat?

DomCR, 2024-10-29:
After the version 2.0.1 the image references were added, that may be the cause of the problem.

Does ARES provide any kind of log about the error? I can only test the file with Autocad on my end.

ChandrasekarML, 2024-10-30:
ARES Commander Log in the command:

Info:Recover dwg file.
Error:Dwg file: DictionaryWithDefault (ACAD_PLOTSTYLENAME) should be Created (Test:)
Error:Dwg file: PrintStyle Normal should be Created (Test:)
Info:Total objects in the handle table found 59.
Info:Loaded objects 59 (with errors 0). Invalid objects 0.
Info:Total errors found during recover: 2.
Info:Check recovered database.
Error:Database header: System Variable "CMLSTYLE" should be Standard (Test:Invalid)
Error:Database header: System Variable "FACETRES" should be Set to 0.5 (Test:Not 0.01..10)
Error:TextStyleTableRecord(31): Is not dependent on a Reference, but ReferenceSymbolId ((4)) is not Null should be Set to Null (Test:Invalid)
Error:Dictionary(2A): Dictionary (ACAD_FIELDLIST) should be Removed (Test:FieldList)
Error:Dictionary(A): Entry object (2A) should be Removed (Test:Invalid)
Error:Layout(34): should be Set to True (Test:Invalid)
Error:MlineStyle(13): Segments amount 0 should be Set to 1 (Test:Not 1..16)
Info:65 objects checked
Info:Total errors found during check 7, fixed 7

Please try with ARES Commander any trial version.

DomCR, 2024-10-30:
Seems that ARES is complaining about some default values and entries in the root dictionary that are not found.

I'll create a branch to fix the ones that appear on the log.

In the meantime it will help if you do this in your end and check if the log improves, for example, to fix the variable FACETRES I think is enough to make sure that is set to a value between 0.01 to 10.0.

There is one element that I don't recognize which is PrintStyle, not sure if it's the same as plotsettings.

Did you try to import a dxf instead of dwg to test if it has the same errors?

ChandrasekarML, 2024-11-12:
We try with dxfwriter, but we get an error. While opening the output drawing file, the drawing file requires a recovery notification to appear.

Sample code:

public void DrawLine(string filePath)
{
    string dwgFilePath = Path.Combine(filePath, "Test.dxf");

    CadDocument doc = new CadDocument();

    void OnNotification(object sender, NotificationEventArgs e)
    {
        Console.WriteLine($"Notification: {e.Message}");
    }

    //Create a line from the origin to the point (5, 5, 0)
    Line line = new Line
    {
        StartPoint = CSMath.XYZ.Zero,
        EndPoint = new CSMath.XYZ(5, 5, 0)
    };

    //doc.Entities.Add(pt);
    doc.ModelSpace.Entities.Add(line);

    doc.Header.Version = ACadVersion.AC1032;

    //using (DwgWriter writer = new DwgWriter(dwgFilePath, doc))
    //{
    //    writer.OnNotification += OnNotification;
    //    writer.Write();
    //}
    using (DxfWriter writer = new DxfWriter(dwgFilePath, doc, true))
    {
        writer.Write();
    }
}
[DrawLine.zip](https://github.com/user-attachments/files/17716249/DrawLine.zip)

DomCR, 2024-11-13:
The issue is the same as dwg, ARES is complaining about the default values.

Have you tried to change them as ARES says before saving the document?

ChandrasekarML, 2025-01-02:
Any update for this query? The drawing file requires recovery notification to appear.

DomCR, 2025-01-02:
Hi @ChandrasekarML,

The current branch has changes that should fix some of the errors that Ares is logging.

Can you send the log that you get when recovering the file using the latest version?

If you are using the repo and not the package, try changing the branch to the linked in this issue and check if the log is different.

Let's see if we can fix this.

ChandrasekarML, 2025-01-08:
Please find the below log file.
[history.txt](https://github.com/user-attachments/files/18341867/history.txt)

DomCR, 2025-01-08:
> ARES Commander Log in the command:
>
> Info:Recover dwg file. Error:Dwg file: DictionaryWithDefault (ACAD_PLOTSTYLENAME) should be Created (Test:) Error:Dwg file: PrintStyle Normal should be Created [...]

This is the log that I need to see what Ares is complaining about.

ChandrasekarML, 2025-01-16:
how to insert an image within a CAD drawing.

please find the below sample code What you tried. Kindly share any sample code on how to insert the image.

            ImageDefinition definition = new ImageDefinition();
            definition.Size = new XY(1, 1);
            definition.Name = "image";

            definition.FileName = Path.Combine(debugPath, "image.jpg");

            RasterImage raster = new RasterImage(definition);

            raster.ClipBoundaryVertices.Add(new XY(0, 0));
            raster.ClipBoundaryVertices.Add(new XY(0, 1));
            raster.ClipBoundaryVertices.Add(new XY(1, 1));
            raster.ClipBoundaryVertices.Add(new XY(1, 0));

            blockDefinition.Entities.Add(raster);

DomCR, 2025-01-16:
Hi @ChandrasekarML,

Your code is correct, only has a few missing parts:

```C#
ImageDefinition definition = new ImageDefinition();
definition.Size = new XY(1, 1);
definition.Name = "image";
definition.IsLoaded = true;

definition.FileName = Path.Combine(debugPath, "image.jpg");

RasterImage raster = new RasterImage(definition);

raster.Flags = ImageDisplayFlags.ShowImage;

raster.ClipBoundaryVertices.Add(new XY(0, 0));
raster.ClipBoundaryVertices.Add(new XY(0, 1));
raster.ClipBoundaryVertices.Add(new XY(1, 1));
raster.ClipBoundaryVertices.Add(new XY(1, 0));

this.Document.Entities.Add(raster);
```

I still have to do some work regarding linking images into in dwg.

Hope this helps.

ChandrasekarML, 2025-01-20:
When will I get the updated version for linking images into DWG?

DomCR, 2025-01-21:
Hi @ChandrasekarML,

Did the code that I send to you work? 

The update will be a utility improvement, right now I'm working on the `XData` features of the library, I'll open a branch and take care of it as soon as I can.

Note: please keep the topic of the issue at hand, if you need to talk/request anything else open a new one, thanks.

DomCR, 2025-01-22:
Keep in mind that the path of the image is relative to the cad file, so if your path is ./image.jpg it means that the image must be in the same folder as you dwg file.

ChandrasekarML, 2025-01-29:
Yes, it's the same path we are referring to, but the image is missing in the DWG file.

Please find the below same code as what we tried.

using ACadSharp;
using ACadSharp.Entities;
using ACadSharp.IO;
using ACadSharp.Objects;
using ACadSharp.Tables;
using CSMath;
using System;

class Program
{
    static void Main(string[] args)
    {
        
        string dwgFilePath = "C:\\Users\\balaji.t\\VIDEOS\\INPUT5\\OUTPUT\\1.dwg";
        string dwgFilePath1 = "C:\\Users\\balaji.t\\VIDEOS\\INPUT5\\OUTPUT\\2.dwg";

        DwgReader reader = new DwgReader(dwgFilePath);

        CadDocument doc = reader.Read();

        void OnNotification(object sender, NotificationEventArgs e)
        {
            Console.WriteLine($"Notification: {e.Message}");
        }
  
        ImageDefinition definition = new ImageDefinition();
        definition.Size = new XY(4608, 3072);
        definition.Name = "123";
        definition.FileName = "C:\\Users\\balaji.t\\VIDEOS\\INPUT5\\OUTPUT\\123.JPG";

        // Add definition to the document blocks
        doc.ImageDefinitions.Add(definition);

        RasterImage raster = new RasterImage(definition);
        raster.Flags = ImageDisplayFlags.ShowImage;
        raster.Size = new XY(4608, 3072); // Set insertion point

        // Correct the clip boundary to match the image dimensions
        raster.ClipBoundaryVertices.Clear();
        raster.ClipBoundaryVertices.Add(new XY(0, 0));
        raster.ClipBoundaryVertices.Add(new XY(0, 3072));
        raster.ClipBoundaryVertices.Add(new XY(4608, 3072));
        raster.ClipBoundaryVertices.Add(new XY(4608, 0));

        doc.Entities.Add(raster);

        using (DwgWriter writer = new DwgWriter(dwgFilePath1, doc))
        {
            writer.OnNotification += OnNotification;

            try
            {
                writer.Write();
                Console.WriteLine($"DWG file successfully written to: {dwgFilePath}");
            }
            catch (Exception ex)
            {
                Console.WriteLine($"Error writing DWG file: {ex.Message}");
            }
        }
    }
}

DomCR, 2025-01-29:
Make sure that the flag isLoaded is set to true:

`definition.IsLoaded = true;`

ChandrasekarML, 2025-02-17:
"While writing the image in the lower version (AC1015), the image is visible in the ARES Commander CAD application. However, when writing the same image in the higher version (AC1032), the image is not visible."

closed 2026-08-19 by DomCR