DomCR/ACadSharp #475: Overwrite Existing File Option

thaiguer, 2024-10-18:
Hello.
The library is very great so far. Congrats @DomCR, and every other contributor.

An option to overwrite the existing file, is desirable.
There should be something like this:

```csharp
using(DwgWriter writer = new DwgWriter(file, doc))
{
   writer.Write(true);
}
```

In the constructor:
```csharp
public override void Write(bool override = false)
```

**Alternative**
Right now, it is necessary to check if the file already exists, and handle it.

DJGosnell, 2024-10-18:
I think that is something that the FileStream should handle and not the writer.
```csharp
FileStream fs = File.Create(path)
```

DomCR, 2024-10-18:
Hi @thaiguer,

I don't quite understand the purpose of this feature, if the goal is to write into the same file you only need to make sure that you dispose the Reader/Writer that you are using to be able to access the file again without triggering an exception, here is an example of how I would implement the overwrite into the same file:

```C#
	CadDocument doc;
	using (DwgReader reader = new DwgReader(_file))
	{
		doc = reader.Read();
	}

	//Do some modifications

	using (DwgWriter writer = new DwgWriter(_file, doc))
	{
		writer.Write();
	}
```

If you test this code with one file you will notice that the changes are applied and you haven't created a new one, the file is overwritten by the writer.

The implementation that @DJGosnell has proposes is actually what the Writer does when you pass a path as a parameter in the constructor, it creates the file and overwrites what's in it.

Let me know if this is the implementation that you are looking for or I misunderstood your proposal.

thaiguer, 2024-10-18:
Thanks.
There is no need to change anything in the library.

My code was getting an exception. But just like @DomCR said: "you only need to make sure that you dispose the Reader/Writer that you are using to be able to access the file again without triggering an exception".

My app was crashing, but the case was probably that other app was with the dwg oppened 😅. My bad, sorry. So, normal behavior. I tested with the code above, and it's overwriting the file with success, as long no other application is using it.

closed 2024-10-18 by thaiguer