DomCR/ACadSharp #482: Throws error "Version not recognized"

donnyv, 2024-10-24:
**Describe the bug**
Throws error "Version not recognized"

**To Reproduce**
Try to load attached file

**Expected behavior**
To load the dxf successfully

[1352_000_01.zip](https://github.com/user-attachments/files/17507453/1352_000_01.zip)

DomCR, 2024-10-24:
Hi @donnyv

I've tested the file and seems to work fine, which version are you using?

donnyv, 2024-12-04:
I just tried the newest version with this file.
[0720_003.zip](https://github.com/user-attachments/files/18011843/0720_003.zip)

Keep getting this error.
[image]

```
ACadSharp.Exceptions.CadNotSupportedException
  HResult=0x80131515
  Message=File version not recognised
  Source=ACadSharp
  StackTrace:
   at ACadSharp.IO.DWG.DwgFileHeader.CreateFileHeader(ACadVersion version)
   at ACadSharp.IO.DwgReader.readFileHeader()
   at ACadSharp.IO.DwgReader.Read()
   at ACadSharp.IO.DwgReader.Read(String filename, DwgReaderConfiguration configuration, NotificationEventHandler notification)
   at ACadSharp.IO.DwgReader.Read(String filename, NotificationEventHandler notification)
   at AcadSharpTest.Form1.button1_Click(Object sender, EventArgs e) in C:\Users\dvela\source\repos\AcadSharpTest\AcadSharpTest\Form1.cs:line 19
   at System.Windows.Forms.Button.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ButtonBase.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(HWND hWnd, MessageId msg, WPARAM wparam, LPARAM lparam)
```
I know this dxf works because I used [netDxf](https://github.com/haplokuon/netDxf) to open it and read all the line work out of it.

Does this library have a dependency that I'm missing?

This the basic project I used to try to open it.
[AcadSharpTest.zip](https://github.com/user-attachments/files/18011936/AcadSharpTest.zip)

DomCR, 2024-12-05:
Hi @donnyv,

You are using the `DwgReader` for a dxf file, please try to use `DxfReader` instead.

Hope this helps.

donnyv, 2024-12-05:
I can't believe I did such a noob move 😞 

Thanks 👍