DomCR/ACadSharp #495: Document.Layouts returning no layouts

ITSJORDAN-TFA, 2024-11-22:
**Describe the bug**
When reading a DWG AC2014 created from DraftSight, it is returning document.Layouts.Count as 0, when it should be 2. 

**Additional context**
Add any other context about the problem here.
If the issue is related to an specific dwg/dxf file and if able, attach that file to the issue of an equivalent with the same error.
I cannot attach the file due to data restrictions but could there be anything that happens in DraftSight that causes this DWG to be mismsatched? Other DWGs from draftsight work fine though

```
public bool HasReadableLayouts()
        {
            using (DwgReader reader = new DwgReader(filePath))
            {
                var document = reader.Read();
                if (document.Layouts.Count() == 0) return false;
                return true;
       }
}
```

DomCR, 2024-11-22:
Hi @ITSJORDAN-TFA,

Older version like AC1014, have an object called VP_ENT_HDR_CTRL_OBJ, which stores the layout and viewport information. This object hasn't been implemented yet, I'll open a branch to fix it.

Thanks for the report!

ITSJORDAN-TFA, 2024-11-24:
Thanks @DomCR, I really love this library!

I will keep an eye out on the branch as I'm curious how you work on these sort of things, the file type is a black box for me

DomCR, 2024-12-19:
I've started to take a look at this one, if you are curious about the dwg format check:

https://www.opendesign.com/files/guestdownloads/OpenDesign_Specification_for_.dwg_files.pdf

closed 2024-12-19 by DomCR

ITSJORDAN-TFA, 2024-12-21:
Appreciate you resolving this I'll rerun my document through, question too. How easy would it be for me to adapt this to be able to read the layout information of the older DWG formats too?

DomCR, 2024-12-21:
I've tested older version of dwg, your case seems to be an exception.

If it doesn't work, could you provide a file as an example so I can test it?