DomCR/ACadSharp #498: Can't set system variable CECOLOR to ByLayer

heatgithub, 2024-12-01:
I wan't to set the system variable CECOLOR to ByLayer otherwise it will be ByBlock in a new dwg file.

I try with

```
cadDoc.Header.SetValue("$CECOLOR", Color.ByLayer);
```

but it's still ByBlock when I open the dwg file.

These two works as expected:

```
cadDoc.Header.SetValue("$CECOLOR", new Color(2));
cadDoc.Header.SetValue("$CECOLOR", new Color(12, 122, 222));
```

Doesn't work with INTERFERECOLOR either, så maybe something with Color.ByLayer?

heatgithub, 2024-12-01:
Found a closed issue 407 about this.
I was using NuGet version 2.4.0-beta and changing to the newer 1.0.2 version solved the issue.
(I don't really understand why 1.0.2 is newer than 2.4.0-beta)

closed 2024-12-01 by heatgithub

DomCR, 2024-12-01:
I've unlisted all pre-release versions to make sure that this issue doesn't happen.