DomCR/ACadSharp #501: Read DimensionAligned bug

SangIk-Lee, 2024-12-06:
**Describe the bug**

DIMENSION_LINEAR recognizes the type DimensionAligned.

SangIk-Lee, 2024-12-06:
When saving a DWG file, dimension objects are not visible. If you select all objects using Ctrl + A, the objects are selected, but the dimension objects' properties are not saved correctly. This issue seems to require correction as well.

DomCR, 2024-12-15:
Hi @SangIk-Lee,

I'll apply the change to the reader, thanks for the report.

I'll need to investigate about the writing problem, I think is related to the block associated to the dimension.

DomCR, 2024-12-15:
What version of the library are you using?

What dimensions are you writing?

This is what I got by adding a simple dimension to a file:
[image]

SangIk-Lee, 2024-12-16:
I did not draw the dimension objects myself but instead used AutoCAD to create 'DimensionLinear' and 'DimensionAligned' in a DWG file. However, when I read the file using DwgReader and saved it using DwgWriter, it did not work properly.

[image]

closed 2024-12-16 by DomCR