DomCR/ACadSharp #512: Modelspace Zoom Extents

sigster, 2024-12-18:
Hi

How can I Zoom Extents in Modelspace  after I import data

sungaoyong, 2024-12-25:
my code worked successed with 
 CadDocument doc = new CadDocument();

 VPort vport = doc.VPorts["*Active"];
            if (vport != null)
            {
                vport.Center = new CSMath.XY(40404219.63209335, 3615454.9612060254);
                vport.ViewHeight = 1202.0730586005643;
                vport.ViewMode = ViewModeType.FrontClippingZ;
                vport.UseDefaultLighting = true;
            }
but not success with CadDocument from a file

DomCR, 2025-01-08:
Hi @sigster 

Haven't found a proper way to do it easily, but I think that the trick should be to play with the `Entity.GetBoundingBox()` and set the `VPort` *Active values to fit with the entities boxes.

The `BoundingBox` struct also has methods like Merge(), which you can use to merge different boxes created form entities and have the total boundaries of your drawing.