DomCR/ACadSharp #514: Reading BlockVisibilityParameters

mme1950, 2024-12-21:
We are reading a dynamic block with a "flip-grip".

The `BlockVisibilityParameters` object associated with the block record specifies a list entities that shall be visible for each visibility state.
When there is a flip-grip another list of handles pointing to some objects representing the grip etc. is added.

In our example three handles apperar referring to objects of the following types. We do not have any documentation about these types.
*  BLOCKFLIPPARAMETER
*  BLOCKFLIPGRIP
*  BLOCKFLIPACTION

The number of entities of this list is read but ignored in the current implementation of the reader for `BlockVisibilityParameters`.
If this number is non-zero the respective number of handles must be read. As long we do not read the object types that are referenced by these handles, these entries can be ignored.

We will fix this issue on open a PR.

DomCR, 2024-12-21:
Hi @mme1950,

I've been looking at the dynamic block properties, here is PR:

- https://github.com/DomCR/ACadSharp/pull/486

I've changed some of the properties on the class that you added, but I think it gives a better structure to read the dynamic parameters.

mme1950, 2024-12-21:
I am a little bit confused. Did we forget to close #206? It should be obsolete after PR #419.

mme1950, 2024-12-21:
Hi @DomCR ,
great - there is a lot of new stuff for dynamic blocks.
Do you have some documentation about dynamic-blocks internals?

mme1950, 2024-12-21:
Hi @DomCR
The new class `BlockVisibilityParameter.State` renamed from `BlockVisibilityParameter.SubBlock` has the additional property `Expressions`. This list will receive references to the three objects mentioned above. So my problem will be solved.

BlockFlipParameter -> Block2PtParameter - > BlockParameter -> BlockElement -> EvaluationExpression
BlockFlipGrip -> BlockGrip -> BlockElement -> EvaluationExpression
BlockAction -> BlockElement -> EvaluationExpression

DomCR, 2024-12-22:
> I am a little bit confused. Did we forget to close https://github.com/DomCR/ACadSharp/issues/206? It should be obsolete after PR https://github.com/DomCR/ACadSharp/pull/419.

Correct, the issue can be closed, the current PR is an improvement over your existing work.

> Do you have some documentation about dynamic-blocks internals?

I don't have any documentation besides the Open design pdf, I've saved a file with a simple dynamic block and controlled values, and I compare the dxf with the values that I can get from the dwg.

For the classes inheritance I'm using the dxf format as an example, it is simpler to do it this way and gives away some clues of how the information is stored. A lot of fields are stored in the exact same order and the `DxfCode` before the value gives a clue of which type can you expect.

mme1950, 2024-12-23:
Hi @DomCR,
this seems to be a quite efficient way to reengineer the EvaluationExpression classes.
We could even implement readers.
Do you think it is possible to evaluate the current state of a used dynamic block in a document?

mme1950, 2024-12-23:
@DomCR BTW: Merry Christmes!

mme1950, 2024-12-23:
Hi @DomCR,
please have a look to this piece of formatted DXF:

```
0   S BLOCKFLIPPARAMETER
5	H	7F959
330	H	7F953
100	S	AcDbEvalExpr
90	BL	6
98	BL	33
99	BL	329
100	S	AcDbBlockElement
300	S	Umkehren
98	BL	33
99	BL	329
1071	BL	0
100	S	AcDbBlockParameter
280	BS	1
281	BS	0
100	S	AcDbBlock2PtParameter
1010	BD	-0.0000000001218723
1020	BD	50.80278554897086
1030	BD	0.0
1011	BD	-0.0000000001218723
1021	BD	109.8133196764683
1031	BD	0.0
----
170	BS	4
91	BL	7
91	BL	0
91	BL	0
91	BL	0
171	BS	0
172	BS	0
173	BS	0
174	BS	0
177	BS	0
----
100	S	AcDbBlockFlipParameter
305	S	Umkehrstatus1
306	S	
307	S	Nicht umgekehrt
308	S	Umgekehrt
1012	BD	-17.54898980138068
1022	BD	108.2007882010403
1032	BD	0.0
309	S	UpdatedFlip
96	BL	7
1001	S	ACAUTHENVIRON
1010	BD	23.1286989982713
1020	BD	27.67408655057769
1030	BD	0.0
```
The part between the dashed lines may have to be added to the `Block2PtParameter` class. The 170 seems to be the number of following 91s.

mme1950, 2025-01-05:
#486 merged, should be solved.

closed 2025-01-05 by mme1950