DomCR/ACadSharp #515: Failed to read dwg file (on nuget version 1.3 or 1.4

enr1qo, 2024-12-23:
I create simple console app for read dwg files to check blocks attributes.
On nuget version 1.2 all worked excelent, but after i upgrade version to 1.3 I have exception -
system.invalidoperationexception collection was modified.
On version 1.4 i get the same error.

Version 1.2 I read 2 entities from dwg and all work fine:
[image]

Version 1.3-1.4 i have an exception:
[image]

But at the same time DxfReader works correctly on all versions.

DomCR, 2024-12-23:
Hi @enr1qo,

Can you provide the file to run the example on my end?

enr1qo, 2024-12-23:
Hi @DomCR !
Here the file.

[test2.zip](https://github.com/user-attachments/files/18230432/test2.zip)

DomCR, 2024-12-23:
I think that the problem is in your code, in the last image you are using the `DwgReader` without disposing it, this may keep the collection busy and throws the exception when you try to access it.

Try to change the code and make it the same as the first image, or use the `DwgReader` like so:

```C#
	CadDocument doc;
	using (DwgReader reader = new DwgReader(_file))
	{
		doc = reader.Read();
	}
```

Hope this helps.

enr1qo, 2024-12-23:
In my old project (on version 1.2) i used using and it worked fine. When i update version, i get an exception, but i think, error was in my project, then i created clear another one.

I change code, but issue still here:

[image]

DomCR, 2024-12-23:
Can you provide the full stack trace of the exception? translated to English please.

enr1qo, 2024-12-24:
Hi @DomCR !
Here the stack trace:

```
System.InvalidOperationException
  HResult=0x80131509
  Message = Collection was modified, enumeration operation may not execute.
  Source = mscorlib
  Stack trace:
   at System.ThrowHelper.ThrowInvalidOperationException(ExceptionResource resource)
   at System.Collections.Generic.Dictionary`2.Enumerator.MoveNext()
   at ACadSharp.IO.Templates.BlockVisibilityParameterTemplate.Build(CadDocumentBuilder builder)
   at ACadSharp.IO.CadDocumentBuilder.BuildDocument()
   at ACadSharp.IO.DWG.DwgDocumentBuilder.BuildDocument()
   at ACadSharp.IO.DwgReader.Read()
   at AcadSharpTest_new.Program.Main(String[] args) in C:\dev\AcadSharpTest_new\Program.cs:line 18
```

DomCR, 2024-12-24:
The `BlockVisibilityParameterTemplate` has a bug when working with a dictionary, the issue is related to a limitation of `NetFramework`, I'll open a PR to fix it.

closed 2024-12-26 by DomCR