DomCR/ACadSharp #524: Read Hatch Edge CircularArc wrong coordinates

gleeuwdrent, 2025-01-10:
**Bug description**
I am reading a DWG with a hatch which is defined with Edges and contains CircularArc. If I create coordinates from the start, mid (halfway) and end point of the circular arcs, they do not always correspond with the geometry of the arc in the DWG.
It looks like the angles of the arc are not correct.

**To Reproduce**
I use this code to print the coordinates:
`foreach(var path in hatch.Paths)
{
	foreach(Hatch.BoundaryPath.Edge item in path.Edges)
	{
		if (item.Type == Hatch.BoundaryPath.EdgeType.CircularArc)
		{
			Hatch.BoundaryPath.Arc arc = (Hatch.BoundaryPath.Arc) item;
			XY startPoint = CalculatePoint(arc.StartAngle, arc.Radius, arc.Center);
			XY midPoint = CalculatePoint(arc.StartAngle + 0.5 * (arc.EndAngle - arc.StartAngle), arc.Radius, arc.Center);
			XY endPoint = CalculatePoint(arc.EndAngle, arc.Radius, arc.Center);
			Console.WriteLine($ "POINT {startPoint}");
			Console.WriteLine($ "POINT {midPoint}");
			Console.WriteLine($ "POINT {endPoint}");
		}
	}
}`

And this method to calculate coordinate from arc properties:
`public static XY CalculatePoint(double angle, double radius, XY center)
        {
            double x = center.X + radius * Math.Cos(angle);
            double y = center.Y + radius * Math.Sin(angle);
            return new XY(x, y);
        }`

**Screenshots**
If I create points from the coordinates in AutoCAD, you can see that some of the arcs are recreated well, but some are wrong.
[image]

**Additional context**
See attached drawing for the dwg.
[hatch.zip](https://github.com/user-attachments/files/18379321/hatch.zip)

DomCR, 2025-01-11:
Hi @gleeuwdrent,

The problem is not with the data, is how is stored and changes depending on the flag `CounterClockWise`.

I've found that the proper way to convert the boundary arcs into `Entities.Arc` is using the following method:

```C#
public ACadSharp.Entities.Arc ConvertToArc()
{
	if (this.CounterClockWise)
	{
		return new ACadSharp.Entities.Arc
		{
			Center = (XYZ)this.Center,
			Radius = this.Radius,
			StartAngle = this.StartAngle,
			EndAngle = this.EndAngle
		};
	}

	return new ACadSharp.Entities.Arc
	{
		Center = (XYZ)this.Center,
		Radius = this.Radius,
		StartAngle = 2 * Math.PI - this.EndAngle,
		EndAngle = 2 * Math.PI - this.StartAngle
	};
}
```

The Arc entity has the method `PolygonalVertexes` which you can use to get the 3 points you need by setting the precision to 3.

I'll add the method to the library and the other boundaries so its easier to get the equivalent entities from it.

Thanks for the report, let me know if this helps.

gleeuwdrent, 2025-01-14:
Thanks, this solves my issue!

closed 2025-01-14 by gleeuwdrent