DomCR/ACadSharp #528: Add Spline constructor

ChrisClems, 2025-01-13:
I'm trying to use ACadSharp to import and export DXF entities to/from a CAD library. I need to convert a spline object from my CAD library type to an ACadSharp Spline entity so that I can create a spline in the CADDocument for export. Unless I'm missing something, there's no way to create a Spline.

I would like a constructor added to the Spline entity class that supports NURBS and fit splines. I'm currently using netDxf which is no longer maintained.

ChrisClems, 2025-01-13:
Nevermind. I see I can build a spline by adding control points, etc. to a new spline object in a loop.

closed 2025-01-13 by ChrisClems

JorenPelfrene, 2025-03-04:
@ChrisClems: is it possible to include a small code example? I'm trying to do something similar.

ChrisClems, 2025-03-04:
> [@ChrisClems](https://github.com/ChrisClems): is it possible to include a small code example? I'm trying to do something similar.

Sure. Here's how I implemented it. The bspline is a spline object from another library. I'm just iterating through the spline data and adding it to the relevant collections on the new AcadSharp Spline object.

```c#
            Spline spline = new Spline();

            if (bspline.ThroughPointCount > 0)
            {
                foreach (GeoPoint fitPoint in bspline.ThroughPoint)
                {
                    spline.FitPoints.Add(new XYZ(fitPoint.x, fitPoint.y, fitPoint.z));
                }

            }
            foreach (GeoPoint controlPoint in bspline.Poles)
            {
                spline.ControlPoints.Add(new XYZ(controlPoint.x, controlPoint.y, controlPoint.z));
            }

            foreach (double weight in bspline.Weights)
            {
                spline.Weights.Add(weight);
            }

            for (int i = 0; i < bspline.Knots.Length; i++)
            {
                for (int j = 0; j < bspline.Multiplicities[i]; j++) spline.Knots.Add(bspline.Knots[i]);
            }
            
            spline.Degree = bspline.Degree;

            return spline;
```