DomCR/ACadSharp #530: The angle of rotation obtained is incorrect

watoken, 2025-01-15:
How to get the correct angle of rotation

DomCR, 2025-01-16:
Hi @watoken,

I guess that you need the angle rotation for a `Text` or a `MText`, they both have the property `Rotation` that will give you the angle, keep in mind that the angle is stored in radians.

Hope this helps.

thaiguer, 2025-04-20:
Hello, @DomCR, and @watoken  . Here is a simple code example on how you can extract the angle rotation of some TEXT and MTEXT objects:

```
using ACadSharp.IO;
using ACadSharp;

string path = @"c:\dev\texts_rotation_sample.dwg";
CadDocument doc = DwgReader.Read(path);

foreach(var obj in doc.Entities)
{
    double rotation = 0;
    string textContent = string.Empty;

    if(obj.ObjectType == ObjectType.TEXT)
    {
        var text = (ACadSharp.Entities.TextEntity)obj;
        double rotationInRadians = text.Rotation;
        rotation = GetRotationInDegrees(rotationInRadians);
        textContent = text.Value;
    }

    if (obj.ObjectType == ObjectType.MTEXT)
    {
        var text = (ACadSharp.Entities.MText)obj;
        double rotationInRadians = text.Rotation;
        rotation = GetRotationInDegrees(rotationInRadians);
        textContent = text.Value;
    }

    Console.WriteLine($"Rotation:{rotation.ToString()} Content:{textContent}");
}

double GetRotationInDegrees(double rotationInRadians)
{
    return rotationInRadians * (180 / Math.PI);
}
```

@watoken , just pay attention that Autocad show the rotation in degrees (0 to 360), and AcadSharp reads it as radians (0 to 6,28). So, a conversion is needed.

[image]

But, @DomCR, **it seemns that AcadSharp is reading always 0 from the property Rotation of MTEXT object**.

File Created with Autocad:
[image]

Result from the example code:
[image]

File Used:
[texts_rotation_sample.zip](https://github.com/user-attachments/files/19825972/texts_rotation_sample.zip)

DomCR, 2025-04-29:
Sorry for the delay, just a quick question, if you save the file the resulting text is rotated? or is in a horizontal line?
Can you check the angle of the Alignment point with the Insert point? maybe for the MText the rotation is not used.

DomCR, 2025-04-29:
AlignmentPoint: {0.7071067811865476,0.7071067811865476,0} //45 degrees
Rotation: 0

I've also noticed that the rotation is not read in dwg, which means that is an optional value and is calculated using the AlignmentPoint.

DomCR, 2025-04-29:
Related PRs:
- https://github.com/DomCR/ACadSharp/pull/88
- https://github.com/DomCR/ACadSharp/pull/82 (needs to be re-applied if rotation == alignment point)
- https://github.com/DomCR/ACadSharp/pull/203

closed 2025-04-29 by DomCR

thaiguer, 2025-04-30:
[image]
It's working.
The merged PRs solved it.