DomCR/ACadSharp #534: Prevent writing file with xData which has an invalid AppId

ChrisClems, 2025-01-16:
Currently ACadSharp will happily export xData with an AppId that hasn't been added to the AppIdsTable, resulting in an invalid DXF. The errors when opening the DXF in AutoCAD are cryptic as it simply says that an entity ended prematurely.

I propose one of two solutions. An exception should be thrown by the writer when attempting to write xData to a file when the xData AppId is not in the AppIdsTable, with a clear message to indicate why the resulting DXF would be invalid. Alternatively, in the Entity.ExtendedData.Add() method, the AppId could be added to the table automatically if the passed AppId is not in the AppIdTable of the owning document. If desired, this could be controlled with an optional bool to preserve the original behavior if someone has a use case for writing xData before the AppId is added to the table.

DomCR, 2025-01-16:
Hi @ChrisClems,

I've been rushing the XData functionality for some time, a fair amount of users where asking for it so I wanted to have something functional right away.

You are right, the current state of the XData is quite unstable, the proper way to do it would be to add the AppId in the table if it doesn't exist when creating the XData. A similar case would be the layers or linetype in the entities.

Thanks for the heads up.

ChrisClems, 2025-01-17:
> Hi [@ChrisClems](https://github.com/ChrisClems),
>
> I've been rushing the XData functionality for some time, a fair amount of users where asking for it so I wanted to have something functional right [...]

Sounds good! I have a small mountain of tests and quirks to get through integrating this into my project, but if I get to a point where I can make the time I will submit a PR for this.

closed 2025-01-20 by DomCR