DomCR/ACadSharp #535: TextEntity need add AdjustAlignment function

sungaoyong, 2025-01-17:
Hello, I used your library to do a transfer of cad documents, the library is very powerful, to solve my needs, but there is a shortcoming to be improved.
My job involves output TextEntity, in which I set up InsertPoint, AlignmentPoint, and alignments such as HorizontalAlignment and VerticalAlignment.
But in fact this involves to recalculate InsertPoint position, for example, I if the horizontal is TextHorizontalAlignment Middle, so is need according to the text AlignmentPoint and output width, Subtract half the width of InsertPoint from the text output width, otherwise the dwg output will display incorrect loading. After loading, the position will be adjusted according to the alignment mode only after the corresponding attribute is refreshed in autocad. Now I write a method separately and add TextEntity for position calculation. I hope this AdjustAlignment function can also be added in TextEntity of this library. It will calculate the output width and alignment method according to the output value. Align the InsertPoint with the AlignmentPoint
I am now calculating that the font output width is the ttf display width calculated using the font of gdi, which may be incorrect for the built-in shx font.

```
Font font = new Font(" Song typeface ", dHeight);
SizeF size = g.MeasureString(strText, font);
return Math.Truncate(size.Width);
```

AdjustAlignment function adjusts the position of InsertPoint according to the alignment method

DJGosnell, 2025-01-17:
I have done a LOT of work with this and to calculate the points, it requires a full text rendering library (like WPF or SKIA for example) to accurately calculate these points as requested. This library does not have those dependencies and I think that if it did, it would quickly bloat the requirements of the library.

My opinion is that this should be the library consumer's job to get these points calculated as it will explode the scope of this project as text rendering is a deep rabbithole...

DomCR, 2025-01-20:
Hi @sungaoyong,

As @DJGosnell, this may add a lot of dependencies and it will defeat the purpose of a lightweight library focused on dxf and dwg.

I might need to take a look at this issue but in this extension project  [`ACadSharp.Pdf`](https://github.com/DomCR/ACadSharp.Pdf) so I can render the text correctly.