DomCR/ACadSharp #536: DIMASSOC objects

mdmozibur, 2025-01-20:
Is there any plan to implement [DIMASSOC objects](https://help.autodesk.com/view/OARX/2018/ENU/?guid=GUID-C0B96256-A911-4B4D-85E6-EB4AF2C91E27) ?
This object is necessary to know which dimension is associated with which entity.

DomCR, 2025-01-21:
Hi @mdmozibur,

I'll open a branch to implement `Dimassoc` into the library, I couldn't find any documentation regarding dwg so by now it will only be implemented for dxf.

Thanks for the report.

mdmozibur, 2025-01-21:
@DomCR I’ve kinda figured out the dwg read part and am looking into the DWG write. Lets have a meeting on this and implement this?

DomCR, 2025-01-31:
Hi @mdmozibur,

Sorry for the delay, if you have code that is working for you, please feel free to open a PR and I'll take a look.

Thanks!