DomCR/ACadSharp #542: Two “72” of Hatch.BoundaryPath.Polyline are generated.

woodbook666, 2025-01-24:
Hello. Please excuse my poor English.
I have encountered a situation where the DWG output is normal and the DXF output has an error when I open the output file, even though they are the same code.
I tried the following sample code and found out that the cause is that when I output a Hatch with Polyline as Path, it outputs two “72” (if I delete one “72”, it opens normally).
Is this a problem with the way I wrote the code?
I would appreciate it if you could check.
```
CadDocument doc = new CadDocument(ACadVersion.AC1032);
var modelSpace = doc.ModelSpace;

string blockName = Guid.NewGuid().ToString();
var block = new Block(new(blockName));
var blockRecord = new BlockRecord(new(blockName));
var insert = new Insert(blockRecord);
modelSpace.Entities.Add(insert);

var hatch = new Hatch()
{
    Pattern = HatchPattern.Solid,
    Color = new ACadSharp.Color(0, 0, 0),
    IsAssociative = false,
    IsSolid = true,
    PatternType = HatchPatternType.SolidFill,
    IsInvisible = false,
    Style = HatchStyleType.Normal,
};

var path = new Hatch.BoundaryPath
{
    Flags = BoundaryPathFlags.External,
};

path.Edges.Add(new Hatch.BoundaryPath.Polyline()
{
    Vertices = [new(0, 0, 0), new(0, 5, 0), new(5, 5, 0), new(5, 0, 0)],
    IsClosed = true,
});
hatch.Paths.Add(path);

blockRecord.Entities.Add(hatch);

using (DxfWriter writer = new DxfWriter("C:\\testdata\\test.dxf", doc, false))
{
    writer.Write();
    writer.Dispose();
}
```

[image]

DomCR, 2025-01-25:
Hi @woodbook666,

Your code is correct, there is a bug in the `DxfWriter`, I'll open a branch to fix it.

Let me know if you are using the repo or the package so I can release an update.

Thanks for the report!

closed 2025-01-25 by DomCR

woodbook666, 2025-01-25:
I am using the package installed via NuGet.
Thanks for the quick response.