DomCR/ACadSharp #555: Support of Flip expressions in dynamic blocks.

mme1950, 2025-02-11:
We are trying to read dynamic blocks with Flip operations.
Flip operations apply to a state of the dynamic block. We need to extract the "normal" and the "flipped" version of the state.

I think for this purpose it should be sufficient to read entities
* BLOCKFLIPPARAMETER
* BLOCKFLIPACTION

Respective classes can be set up as

    public class BlockFlipParameter : Block2PtParameter {
    }

    public class BlockFlipAction : BlockAction {
    }

I implemented readers and templates for
* `BlockFlipAction`
* `BlockAction`
* `BlockFlipParameter`
* `Block2PtParameter`

The evaluation expression are referenced by the state of the dynamic block. This is nice but it is half the job,

To extract the flipped version of the dynamic-block State we have to apply the flip operation to the entities listed in the `BlockFlipAction` object, i. e. we must mirror all coordinates of the entities at the line defined by the two XYZs provided by the `BlockFlipParameter` object. Texts, however, must not be mirrored, but the angle and alginment has to be changed.

@DomCR Do you have a suggestion how to include such functionality into ACadSharp?

DomCR, 2025-02-12:
Hi @mme1950,

I started working on a Transform utility to make this kind of operations:

- https://github.com/DomCR/ACadSharp/pull/457

The idea was that the user can apply a transformation matrix into any entity.
This feature can be extended for the different `BlockAction` so they generate their own transformation matrix and apply the transformation into the entities in the block.

Note: the mirroring of the text shouldn't be a problem I think is controlled by `TextMirrorFlag`.

mme1950, 2025-02-12:
Hi @DomCR,
this is really great, Thank, you.

DomCR, 2025-11-15:
Looks good! 

Flip transform will act as mirror? or symmetry? don't know which one is more readable in the different Math softwares but it's ok for me.

Have you implemented the method `CreateFlipTransform`?

mme1950, 2025-11-15:
so, as far as I know the FlipAc0tion in AutoCAD is only defined in 2D.
So the implementation fragment above should be sufficient.
Before we have to evaluate the mirrorDir vector as perpendicular to the mirror-axis vector:

`mirrorDir = new XYZ(axisDir.Y, -axisDir.X, 0)
`
The sign is irrelevant because the Matrix4 only contains products of components of `mirrorDir`-Components.

Here is the code:

```
        public static Matrix4 CreateFlipTransform(XYZ pointOnAxis, XYZ axisDirection) {

            var nx = axisDirection.Y;
            var ny = -axisDirection.X;

            Matrix4 mirror = new Matrix4(
                1 - 2 * nx * nx, -2 * nx * ny, 0, 0,
                -2 * ny * nx, 1 - 2 * ny * ny, 0, 0,
                0, 0, 0, 0,
                0, 0, 0, 1
            );

            Matrix4 toOrigin = Matrix4.CreateTranslation(-pointOnAxis.X, -pointOnAxis.Y, -pointOnAxis.Z);
            Matrix4 backFromOrigin = Matrix4.CreateTranslation(pointOnAxis.X, pointOnAxis.Y, pointOnAxis.Z);

            Matrix4 flip = backFromOrigin * mirror * toOrigin;

            return flip;
        }

```
I will open a PR.

closed 2026-08-19 by DomCR