DomCR/ACadSharp #558: DWGWriter thumbnail support

ChrisClems, 2025-02-12:
I want to say before typing up this issue, I know this is nobody's top priority and this is likely a huge amount of work. I'd like to take on the heavy lifting and eventually submit a PR for this if it would be welcome and I can figure it out. I'm starting an issue to 1) see if this functionality could be packaged in a way that would be accepted as a PR in ACadSharp and if so 2) consult the experts to see if they have any information that might save me some time and research.

I process massive amounts of individual AutoCAD files. My library is tens of thousands of drawings and I'm working on up to 1,000 at a time in a single project. It is incredibly useful to me to spot-check thumbnails in folders to see if there are any glaring issues I need to address. I usually export DXF files when I'm working with ACadSharp because I have software on my machine which auto-generates thumbnails for them. It would be beneficial to me to be able to output DWG files with thumbnails.
 
The way I understand it, DWG thumbnails are generated by AutoCAD and saved in the file header in various formats depending on the DWG version. Currently, the only way to generate a DWG with a thumbnail which was not originally saved with one is to open it in AutoCAD and save it. It looks like there's some information on storing BMP and WMF thumbnails on [page 93 of the ODS DWG spec document](https://www.opendesign.com/files/guestdownloads/[OpenDesign_Specification_for_.dwg_files.pdf). There are format changes in R13C3. I think with this document and a bit of work reading how the DwgWriter works I might be able to implement this as an option to pass when writing a DWG file.

I understand the desire to keep this a lightweight library so any sort of rendering dependencies are obviously out. My alternate idea is optionally passing a user-generated byte array representing a BMP to the DwgWriter that would be inserted at the correct spot. There is already some precedent for this as the DwgReader is retrieving what appears to be preview image data as byte arrays when importing BlockRecords.

Or rather than passing a user-generated byte array, I don't think it would be that insane to implement a simple BMP rasterization class. The BMP spec is dead simple and is not exactly going to be undergoing any major breaking changes that would create a maintenance issue for this class. I wrote up a super simple proof of concept that renders vector line and arc data to a BMP and it's working in under 200 lines of code. Obviously expanding that to support more entities would increase the size but I think a BMP raster class could be quite compact. It would only need to support the most basic entities such as lines, arcs, circles, ellipses, polylines, and splines. Polylines can be exploded for the purposes of rasterization to keep things simple. Text, hatching, and other more complex entities are already being ignored in AutoCAD thumbnails so we can as well. Splines are the only thing I'm having difficulty with.

If I can create a working version of DwgWriter that generates valid DWG files with thumbnails and make a reasonably small and basic rasterization class that will create a BMP byte array from primitive entities in model space to pass to it, would you be open to a PR for a new feature?

DomCR, 2025-02-14:
Hi @ChrisClems,

We can take a look at this feature, the reading part is easy, check how the thumbnail is stored and read it as a byte array or even a bitmap.

To write it that's another story, the idea of the user giving the byte array is a good one, if they have the tools to generate the image or want a custom want is just a matter of writing those bytes.

The only rule that I would impose is that if the feature needs some external or third party libraries maybe it would be best to create a separated project like I did with `ACadSharp.Pdf` if more formats are involved maybe I event change the name to `ACadSharp.Media` and add them there.

I'll take a look when I can and see what we can do.

Thanks for your support.

ChrisClems, 2025-02-15:
@DomCR Thanks! So I went a little further into this last week than I'd planned before hearing back from you. I left all my work on this at the office so I won't be able to dig back in until next week but here's a brain dump while it's still fresh in my memory.

I believe I've found a bug in the DwgWriter. This was my initial testing process:

1. I created a BMP byte array and tried to insert it in the preview section based on the linked documentation. I could not get this to work.
2. Digging into some DWGs straight out of AutoCAD, I determined that at least AutoCAD 2018 did not have any way to write anything but PNG thumbnails. I figured maybe since that documentation was published they dropped BMP support so I started focusing on PNGs.
3. I reverse engineered what I believe is the preview section for a 2018 DWG file which is different than the BMP version in the linked doc. There are a few sections where I don't know what they do, but the values are static enough that I don't believe we need to know to make it work. The essential entries are the overall size of the image section and the size of the PNG itself, both represented by a 32 bit UINT.
4. Adding the logic to the DwgPreviewWriter to create the correct code does not work. It creates a seemingly valid file for some versions, but Windows will not pick up the thumbnail.
5. Some versions will write files that when opened will trigger the recovery dialog in AutoCAD. I didn't have a lot of time to dig into this before I left, but I believe AC2018 and AC2032 were working, while others tested were outputting broken files. Both of these specifically use DwgFileHeaderWriterAC18() to write the header. Not sure if that's relevant.
6. In the interest of being thorough, I took a working DWG and I hand-wrote a new preview section in a hex editor using the same PNG data that I was trying to inject via the DwgPreviewWriter and Windows picked it up as soon as I saved the file. I cross-referenced my hand-written preview section with what ACadSharp was putting in the file and they matched. This leads me to believe that the preview section may not be where AutoCAD expects it to be in the file. In a working DWG saved from AutoCAD, the preview section is very close to the beginning of the file. In ACadSharp there's much more data and a bunch of zero padding between the beginning of the file and the preview section.

For what it's worth, I also managed to cobble together a PNG encoder that takes lines, arcs, ellipses, elliptical arcs, fit splines, and NURBS splines, and rasterizes them with zero external dependencies. It's under 1,000 lines of code. I needed some help from Claude to get the NURBS spline approximation working but the experimental code had a really clean interface. The PNGs it outputs are passing `pngcheck` with no issues and the compression is also handled with standard libraries.

I'll do more cross referencing next week with the section layout in my DWG files and see if I can find where the problem is. If I can't figure it out myself by week's end, I'll dump a bunch of examples here so someone else can check my work.

DomCR, 2025-02-16:
> 5. Some versions will write files that when opened will trigger the recovery dialog in AutoCAD. I didn't have a lot of time to dig into this before I left, but I believe AC2018 and AC2032 were [...]

The version AC14 and AC15 are quite old, the writer is implemented but is not super stable and the files usually need to be restored, it also needs to be refined to be able to write an obsolete object, VP_ENT_HDR_CTRL_OBJ which it stores and controls all the viewports.

The version AC21 is a particular one, Autodesk changed some parts of the encoding, but only for this one, I haven't implemented yet, there is a branch created but the amount is quite high and there are some other priorities before implementing a writer for an older version.

AC18, AC27 (recently fixed) and AC32 are the ones that work better and the best to test with, the writer is the most stable.

>6. In a working DWG saved from AutoCAD, the preview section is very close to the beginning of the file. 

This should not matter, in the `FileHeader` there is the Preview offset which I believe that Autocad will get to render the preview in the browser.

For AC15:
https://github.com/DomCR/ACadSharp/blob/bed98289b16469a379f804f4c61c9bea820f1914/src/ACadSharp/IO/DWG/DwgReader.cs#L444

For AC18 and after:
https://github.com/DomCR/ACadSharp/blob/bed98289b16469a379f804f4c61c9bea820f1914/src/ACadSharp/IO/DWG/DwgReader.cs#L953

In the writer for AC18 and after:
https://github.com/DomCR/ACadSharp/blob/bed98289b16469a379f804f4c61c9bea820f1914/src/ACadSharp/IO/DWG/DwgStreamWriters/DwgFileHeaderWriterAC18.cs#L311

> In ACadSharp there's much more data and a bunch of zero padding between the beginning of the file and the preview section.

Thanks a lot for this input ❤! the 0s where related to the DxfClasses section not being compressed, now the files are much lighter too.
I've also dig about the AC27 version not working and manage to fix it. 

- https://github.com/DomCR/ACadSharp/pull/562

DomCR, 2025-02-16:
I've been trying to generate a dxf with a preview image but Autocad doesn't have the feature enabled for some reason, I wanted to try to generate the image from a dxf file which is way more transparent than a dwg.

Where do you create the image preview for your dxf?

DomCR, 2025-02-18:
I managed to make it work using the branch linked to the issue, here is the code to test it:

```C#
			CadDocument doc;
			DwgPreview preview;
			using (DwgReader reader = new DwgReader(_file))
			{
				doc = reader.Read();
				preview = reader.ReadPreview();
			}

			string output = Path.Combine(Path.GetDirectoryName(_file),
				$"{Path.GetFileNameWithoutExtension(_file)}.out.dwg");
			using (DwgWriter writer = new DwgWriter(output, new CadDocument()))
			{
				writer.Preview = preview;
				writer.Write();
			}
```

I'll try to refine the code and fix the offset positioning.

closed 2025-04-01 by DomCR