DomCR/ACadSharp #574: Write Face3D Can't Render in AutoCAD

Lindingdou, 2025-03-06:
[image]
The above code can successfully write a dwg file, but write a three-dimensional surface, when using AutoCAD rendering will appear fatal errors

[image]

[image]

Is this a common problem, or is it caused by my improper operation.

Thank you. I'll be glad if you can help me. Thank you

DomCR, 2025-03-06:
Hi @Lindingdou,

It doesn't seem to be related to the Face3D, your implementation looks good there is no issue there, the problem seems to be the switch to the 3D, I've tried to change the view for any file generated by `ACadSharp` and all of them seem to have the same error for any version after `AC1018`.

Try to save the file in `AC1018` this should solve the issue for now, the needed fix I think is related to the materials, they where introduced after this version and they are not implemented for `ACadSharp`.

I'll open a branch to try to see if I can find out how to read them.

Thanks for the report!

Lindingdou, 2025-03-06:
Thank you very much for your very timely reply. I will try AC1018 again

Lindingdou, 2025-03-09:
Sorry to bother you again, I did solve my problem after switching to AC1018, but I seem to have found a new bug, AC1018 version can not display Chinese, all display contents are??？.The switch to gb2312 did not work, and was resolved after switching to AC1032

DomCR, 2025-03-09:
Here is a similar discussion about this issue:

- https://github.com/DomCR/ACadSharp/issues/432

For version older than AC1018 it seems that the special characters are encoded in a different way, here is what I've read from a file saved in AC1018:

Value: "\\M+2A7DA\\M+2AABA\\M+2B575\\M+2AB48"
Original: 我的短信

I need to investigate some more about this, but it seems that for special characters in older version the text is saved directly in hex.

closed 2025-12-02 by DomCR