DomCR/ACadSharp #575: Cannot Write Group

OndrejJanota91, 2025-03-07:
Hello, I am trying to create a simple DWG/DXF with few entities grouped together. However, If I add 
group with the entities to the document and open document entities are not grouped. If I read the document with the code it contains group with group does not contain entities. Code is bellow. Am I missing something? thanks for reply.

`using ACadSharp;
using ACadSharp.Entities;
using ACadSharp.IO;
using ACadSharp.Objects;
using ACadSharp.Tables;

namespace ACADDXF
{
    class Program
    {
        static void Main(string[] args)
        {
            CadDocument doc = new CadDocument(ACadVersion.AC1032);
            using (DwgWriter writer = new DwgWriter("testDocumnet22.dwg", doc))
            {

                Layer layer = new Layer("MyLayer");
                layer.Color = new Color(0, 153, 0);
                doc.Layers.Add(layer);

                Circle circle = new Circle();
                circle.Center = new CSMath.XYZ(1, 1, 0);
                circle.Radius = 1;
                circle.Normal = new CSMath.XYZ(0, 0, 1);
      
                Line line = new Line();
                line.StartPoint = new CSMath.XYZ(0, 0, 0);
                line.EndPoint = new CSMath.XYZ(2, 2, 0);

                circle.Layer = layer;
                line.Layer = layer;

                doc.Entities.Add(circle);
                doc.Entities.Add(line);

                Group group = new Group();
                group.Name = "MyGroup";
                group.Entities.Add(1, circle);
                group.Entities.Add(2, line);
                group.Selectable = true;

                doc.Groups.Add(group);

                TextEntity text = new TextEntity();
                text.Value = "Hello World!";
                text.Layer = layer;
                text.HorizontalAlignment = TextHorizontalAlignment.Center;
                text.VerticalAlignment = TextVerticalAlignmentType.Middle;
                text.InsertPoint = new CSMath.XYZ(1, 1, 0);
                text.AlignmentPoint = new CSMath.XYZ(10, 10, 0);

                doc.Entities.Add(text);

                writer.Write();

            }

            using (DwgReader reader = new DwgReader("testDocumnet22.dwg"))
            {
                CadDocument doc1 = reader.Read();
                int i = 0;
            }
        }
    }
}
`

DomCR, 2025-03-07:
Hi @OndrejJanota91,

The `Group` is not fully implemented, the entities collection are not fully linked to the `CadDocument` and the handles are wrongly stored.

I'll open a PR to fix it.

Thanks for the report!

OndrejJanota91, 2025-03-10:
Thanks for the reply.

closed 2025-03-12 by DomCR